FPGAwars/icestudio #441: Toolchain installation stall in nightly build on Linux Mint 20

otto9G9otto, 2020-10-17:
The toolchain installation step stalled in nightly build 0.5.1n201009 on Linux Mint 20. The more stable 0.5.0 app image did not run at all.

wwlin1198, 2021-02-18:
For nightly, this is what worked for me. I was stuck at 10% on installing toolchain and it gave me some ensure pip error and others. 
1. Make sure that your pip is up to date and venv installed
2. This should help https://stackoverflow.com/questions/24123150/pyvenv-3-4-returned-non-zero-exit-status-1

qpeters, 2021-07-13:
@otto9G9otto is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

Obijuan, 2022-01-17:
It should work ok in the latest icestudio version (0.8.2w). Feel free to open another issue if the toolchain is not correctly installed

closed 2022-01-17 by Obijuan