FPGAwars/icestudio #444: Error with wine-mono while building icestudio from main branch on Ubuntu 18.

aalku, 2020-10-20:
I get these errors while building icestudio from **main** branch on Ubuntu 18.

I run `npm run dist`

I get:
```
Fatal error: Error while updating the Windows icon. Wine (winehq.org) must be installed to add custom icons from Mac and Linux.

0009:err:mscoree:CLRRuntimeInfo_GetRuntimeHost Wine Mono is not installed
```

I tried:
```
sudo apt-get install wine
sudo apt-get install mono-complete
# and mono-anythingIcanThinkOf
```

cavearr, 2020-10-20:
Hi! for build, you need ubuntu 18.04 with node 10.7 with this conf should be work

aalku, 2020-10-20:
Hi! If that's the case I think that should be in the README but I updated node to the latest 10, 10.22.1 and still doesn't work.

I don't think that's the problem.

[image]

cavearr, 2020-10-20:
You can't built on newer versions of node, only works in node 10.17.0.

And you need to install wine and ncis with:

sudo apt-get install wine nsis

aalku, 2020-10-20:
I think "wine" is not a package name on the Ubuntu 18 official repos.

[image]

I tried wine64.

I found it hard to find node 10.17.0 package but I did it.

https://deb.nodesource.com/node_10.x/pool/main/n/nodejs/nodejs_10.17.0-1nodesource1_amd64.deb

I think it's the official node.js distribution repository for Ubuntu.

[image]

Still the exact same error I shown before.

Maybe the problem is I have to use the mono version that is shown in the error message: winehq.org 🤔

aalku, 2020-10-20:
Yup, that was it. I followed winehq.org instructions for ubuntu 18. There are a lot of steps to install it, with the referenced audio library.

```
sudo dpkg --add-architecture i386
wget -nc https://dl.winehq.org/wine-builds/winehq.key
sudo apt-key add winehq.key
sudo add-apt-repository 'deb https://dl.winehq.org/wine-builds/ubuntu/ bionic main'

wget -nc https://download.opensuse.org/repositories/Emulators:/Wine:/Debian/xUbuntu_18.04/Release.key
sudo apt-key add Release.key
sudo add-apt-repository 'deb https://download.opensuse.org/repositories/Emulators:/Wine:/Debian/xUbuntu_18.04/ ./'
sudo apt update

sudo apt install --install-recommends winehq-stable
```

Now I have a completely different error.

**Don't you think more details about this should be in the README as solution to this issue?**

BTW, I think node version was not the issue in this error. I updated it by mistake and this step still work.

Thank you!

cavearr, 2020-10-21:
hi, i think you have different kind of problems. Documentation is very oldy and deactualiced , i'm trying to explain de process you need to install wine.

First of all, wine is the first. Wine is used for software thar builds Windows executables.

For your message i think you doesnt' have de correct wine version (or packages for wine version).

If you install from winehq, you install the latest wine version that is not compatible with the packages that icestudio build process uses.

Sorry for this but the current stable and developement version of icestudio has a big dependency with obsoleted packages, we are working on it but the new version  is in very early bird state. 

For this , if you want to build icestudio, yo need and "oldy" environment to work with it, i'm recommending you install ubuntu desktop 18.04 in a virtualmachine (probably we probide soon of ova with it) and install on it only the indispensable software.

You need wine version probided by ubuntu 18.04 no the latest version of wine) only with apt-get install wine should be enouth.

For install old node version is very easy, I'm not recommend install global deb packages because you want to use de newest for other.

You  could do in different ways, i prefer download the zip or tgz package  from here https://nodejs.org/dist/v10.17.0/

decompress in a diretory on your home and add to your path the bin directory of this version of node (you can't have other node version globally installed). I have a direcotry in my user account with different versions of node and an script that permits me change between versions in each terminal session i'm opening, you can do as you want.

If you use a virtual environment that  i'm recomending you could install globally because you only use this machine for it.

With this you could build icestudio.

aalku, 2020-10-21:
- Thank you for you explanation. The bad news is that I installed Ubuntu 18.04 (virtual machine), and I installed the node version you said (other way but same version) and I did "sudo apt install wine" and it didn't work.

If you are working on fixing the build system then I understand this is not a problem in the long run.

I'm trying to build icestudio myself because I added a new package for another board to apio and I want to test everything end-to-end with the prepackaged apio system you use before making an icestudio pull request for that board.

The apio change is already merged to the development branch.

Thank you again! Have a nice day!

cavearr, 2020-10-21:
Don't worry, i'm tried to help you.  Please send me the output the command:

cat /etc/apt/sources.list

aalku, 2020-10-21:
This is how it is right now after a lot of try-and-error. I'm sure I added things that were not actually needed.
I got a snapshot of the VM just installed, I can go back to it and start again but my current priority is to build the develop branch, test it on windows, switch apio reference to the apio develop branch, build again, and try the new board on Windows so I can be sure I did everything right and I can pull request the changes. Then I'll add support for other architectures on the board programmer package :)
Thank you.

```
ubuntu@ubuntu:~$ cat /etc/apt/sources.list
# deb cdrom:[Ubuntu 18.04.5 LTS _Bionic Beaver_ - Release amd64 (20200806.1)]/ bionic main restricted

# See http://help.ubuntu.com/community/UpgradeNotes for how to upgrade to
# newer versions of the distribution.
deb http://us.archive.ubuntu.com/ubuntu/ bionic main restricted
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic main restricted

## Major bug fix updates produced after the final release of the
## distribution.
deb http://us.archive.ubuntu.com/ubuntu/ bionic-updates main restricted
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic-updates main restricted

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team. Also, please note that software in universe WILL NOT receive any
## review or updates from the Ubuntu security team.
deb http://us.archive.ubuntu.com/ubuntu/ bionic universe
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic universe
deb http://us.archive.ubuntu.com/ubuntu/ bionic-updates universe
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic-updates universe

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu 
## team, and may not be under a free licence. Please satisfy yourself as to 
## your rights to use the software. Also, please note that software in 
## multiverse WILL NOT receive any review or updates from the Ubuntu
## security team.
deb http://us.archive.ubuntu.com/ubuntu/ bionic multiverse
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic multiverse
deb http://us.archive.ubuntu.com/ubuntu/ bionic-updates multiverse
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic-updates multiverse

## N.B. software from this repository may not have been tested as
## extensively as that contained in the main release, although it includes
## newer versions of some applications which may provide useful features.
## Also, please note that software in backports WILL NOT receive any review
## or updates from the Ubuntu security team.
deb http://us.archive.ubuntu.com/ubuntu/ bionic-backports main restricted universe multiverse
# deb-src http://us.archive.ubuntu.com/ubuntu/ bionic-backports main restricted universe multiverse

## Uncomment the following two lines to add software from Canonical's
## 'partner' repository.
## This software is not part of Ubuntu, but is offered by Canonical and the
## respective vendors as a service to Ubuntu users.
# deb http://archive.canonical.com/ubuntu bionic partner
# deb-src http://archive.canonical.com/ubuntu bionic partner

deb http://security.ubuntu.com/ubuntu bionic-security main restricted
# deb-src http://security.ubuntu.com/ubuntu bionic-security main restricted
deb http://security.ubuntu.com/ubuntu bionic-security universe
# deb-src http://security.ubuntu.com/ubuntu bionic-security universe
deb http://security.ubuntu.com/ubuntu bionic-security multiverse
# deb-src http://security.ubuntu.com/ubuntu bionic-security multiverse
deb https://dl.winehq.org/wine-builds/ubuntu/ bionic main
# deb-src https://dl.winehq.org/wine-builds/ubuntu/ bionic main
deb https://download.opensuse.org/repositories/Emulators:/Wine:/Debian/xUbuntu_18.04/ ./
# deb-src https://download.opensuse.org/repositories/Emulators:/Wine:/Debian/xUbuntu_18.04/ ./
ubuntu@ubuntu:~$ 
```

cavearr, 2020-10-21:
Excuse me! I'm wrong! the correct linux version to build without dependency problems is Ubuntu 16.04 LTS

closed 2020-10-21 by cavearr