FPGAwars/icestudio #445: Error with libgconf-2 while trying to run icestudio from main branch sources on Ubuntu 18.

aalku, 2020-10-20:
I have an error with libgconf-2 while trying to run icestudio from main branch sources on Ubuntu 18.

npm start gets stuck on `Waiting...` and the app does not start.

Examining the logs I find it may be an issue with `nw`. That command I don't have in my system path but that's not the problem.
I find `nw `in `icestudio/node_modules/nw/bin/nw` but running it it seems to lack `libgconf-2.so` or something like it. I don't have the exact error anymore because I found a workarround.

That workarround is `sudo apt install libgconf-2-4`

After that `npm start` starts working.

I don't know how this dependancy issue should be fixed or documented in the installation procedure.

alexcweiner, 2020-10-24:
same thing happens in Ubuntu 20.04.1 LTS. Same workaround works too.

cavearr, 2020-10-24:
Hi alex! What repository are you cloning?

Could you paste the url of the repo?

Thanks!

----- Mensaje original -----

https://github.com/FPGAwars/icestudio/issues/445#issuecomment-715662034

alexcweiner, 2020-10-24:
I used the Linux 64 bit AppImage file available at https://icestudio.io/#lk-download

cavearr, 2020-10-25:
Try Thais command to install the dependency:

sudo apt install libgconf-2-4

And then rin icestudio.

Tell is if it works!

----- Mensaje original -----

https://github.com/FPGAwars/icestudio/issues/445#issuecomment-716068157

Obijuan, 2022-01-17:
I think this bug is fixed and the issue can be closed. Feel free to reopen it you still get the same error

closed 2022-01-17 by Obijuan