FPGAwars/icestudio #446: Error with grunt-appdmg@1.0.0 while running "npm install" on icestudio develop branch on Ubuntu 18.

aalku, 2020-10-20:
I have this error when trying to run `npm install` on icestudio develop branch Ubuntu 18.

```
$ npm install
npm ERR! Linux 5.4.0-52-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code EBADPLATFORM

npm ERR! notsup Not compatible with your operating system or architecture: grunt-appdmg@1.0.0
npm ERR! notsup Valid OS:    darwin
npm ERR! notsup Valid Arch:  any
npm ERR! notsup Actual OS:   linux
npm ERR! notsup Actual Arch: x64
```

aalku, 2020-10-20:
It seems the problem is here. I think `package.json` does not support dependencies by os, so you can either put it as optional dependency and check if it's installed (on darwin) later or use a script that runs `npm install grunt-appdmg` only if os is darwin.

[image]

cavearr, 2020-10-20:
yes this is an error on last commits, i'm fix it, now is correct, thanks for your support!!

closed 2020-10-20 by cavearr