FPGAwars/icestudio #447: I can not update toolchain with version 0.5.1n201009 in Windows10 chinese version

albertlu2013, 2020-10-22:
Hi :
When I update toolchain to apio 0.5.5 in Windows10 Chinese version.  I get the error message :
                                                                                                          
	Cloning https://github.com/FPGAwars/apio.git (to revision develop) to 
c:\users\albert\appdata\local\temp\pip-install-6fv42p2y\apio                                                                                                                ERROR: Command errored out with exit status 1:                                                                           
command: 'c:\users\albert\appdata\local\programs\python\python38\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\albert\\AppData\\Local\\Temp\\pip-install-6fv42p2y\\apio\\setup.py'"'"'; __file__='"'"'C:\\Users\\albert\\AppData\\Local\\Temp\\pip-install-6fv42p2y\\apio\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' 
	egg_info --egg-base 'C:\Users\albert\AppData\Local\Temp\pip-pip-egg-info-_jtgn3u6'                                              
	cwd: C:\Users\albert\AppData\Local\Temp\pip-install-6fv42p2y\apio\                                                 
	Complete output (5 lines):                                                                                              
	Traceback (most recent call last):                                                                                        
	File "<string>", line 1, in <module>                                                                                    
	File "C:\Users\albert\AppData\Local\Temp\pip-install-6fv42p2y\apio\setup.py", line 25, in <module>                        long_description=open('README.md').read(),                                                                          
	UnicodeDecodeError: 'cp950' codec can't decode byte 0xcb in position 654: illegal multibyte sequence                    ----------------------------------------                                                                            
	ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.  

I found "[Fix ‘cp950’ Error when using ‘pip install](https://wesely.github.io/pip,%20python,%20pip/Fix-'cp950'-Error-when-using-'pip-install'/)’". But I can not fix it in my local place.

Would you please help to fix it? Thanks

WiFiBoy, 2021-01-16:
Just modify apio/setup.py line 25: open('README.md', encoding='utf-8').read() and the problem is solved.

Obijuan, 2022-01-17:
As there is no more news, and the latest icestudio version is 0.8.2w, I will close this issue

closed 2022-01-17 by Obijuan