FPGAwars/icestudio #448: Sigrock plugin, error on Windows

aalku, 2020-10-24:
Sigrock plugin is not correcly working on windows.
I debugged two reasons.

1. It's using icestudio folder for the raw file. That's forbidden unless running as administrator. It should use for example `%userprofile%/.icestudio/`
2. It's not supporting blanks in the path to the raw file when calling pulseview.exe. Something's wrong there.

It's working nicelly if I override the default behaviour of windows by installing icestudio on `C:\icestudio` instead of the default `c:\program files\icestudio` and then running icestudio as administrator (equivalent to sudo on Windows).

Just a small issue of a work in progress plugin. Great work!

cavearr, 2020-10-25:
Thanks for test in Windows! I’m checking the spaces in the path for this os.

Anyway In the next days i’m updating icerock and i’m changing the capture.raw fonder to make more accessible.

Thanks!

----- Mensaje original -----

https://github.com/FPGAwars/icestudio/issues/448

cavearr, 2020-11-22:
In develop branch is fixed, icerok.raw creates in the ice-build folder created in the same dir that your .ice file.

If you dont save your .ice file or icestudio cant write in the .ice dir, uses temporal folder.

Try it and tell us.

This night will be a new nightly build if you want to download the binary.

Stay this issue open until you tell us if it works.

Thanks a lot!

aalku, 2020-11-23:
It's not working for me :(
Pulseview should open automatically when I stop the capture. Right? 
It dosn't do it. If it's open it doesn't open the raw file either.

aalku, 2020-11-23:
This is failing for me:

[image]

`this.serial.sm.info.conn` is an object, not a boolean, so `this.serial.connected` isn't set to true so disconnection isn't handled properly later.

It seems it shall never be true, just false or object. 🤔

cavearr, 2020-11-24:
Yes this versión is a probé of concept i am redactoring it, wait some days!! 🙃

qpeters, 2021-07-13:
@aalku is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0