FPGAwars/icestudio #456: [Dev-0.5.1n201120] Conversion window does not get closed

Obijuan, 2020-11-21:
When an example is loaded from the default collection, there appear a windows for converting the example to your board or load as it is. But that windows is never closed, and it is not possible to work with the example

In this animation is shown how to reproduce the bug

[image]

* Icestudio version: nightly 0.5.1n201120
* Operating system: Ubuntu 20.04

cavearr, 2020-11-22:
Fixed error, icestudio can't create the build dir folder (new feature) and hangs. 

Only affect to linux because appImage create a virtual mount point in only read mode.

Is fixed yet, now, when Icestudio can't create the build dir folder in the .ice folder, create it in old way at the os temporal directory.

This issue is the same that #457 

Stay open until you confirm it works

Obijuan, 2020-11-23:
Checked! It works! 😀️  Thanks a lot!

closed 2020-11-23 by Obijuan