FPGAwars/icestudio #461: ULX3S board in windows: zadig should no be launched

Obijuan, 2020-11-22:
The driver for the ulx3s board in Windows does no need to be installed with zadig (that is now automatically executed when the drivers are configured). If the libusbk (from zadig) is installed, the bitstream cannot be uploaded with fujprog

BPJWES, 2021-08-16:
Hi,

I also run into this problem using windows 8.1 in combination with the ULX3S-85F.

I have circumvented this by running fujprog standalone from the command line, however it would be really nice if this would also work from within icestudio on windows.

I have had a cursory look into the icestudio code, and I think this problem actually lies within APIO. 

As this problem is as far as I can tell specific to the ULX3S boards, the APIO function _ftdi_enable_windows()
(https://github.com/FPGAwars/apio/blob/93957521ae12cf20edf3060015935172feb3a686/apio/managers/drivers.py#L405) should somehow get information from the ULX3S board definition that running zadig is unnecessary.

cavearr, 2025-01-08:
this is fixed in the latest wip with apio development toolchain. https://downloads.icestudio.io

closed 2025-01-08 by cavearr