FPGAwars/icestudio #462: Can not install Tollchain on Win 10 devices

ph1lj-6321, 2020-11-25:
This seems to be a problem, it stalls on trying to install the VirtualEnv (10%) into the process.
Python 3.9, 3.8 & 2.7 all installed and in the Path.

Obijuan, 2020-11-25:
We have tested the latest version, icestudio-0.5.1n201120, in different windows 10 machines and it was ok. Let's try to find why it is not working ok in your machines

I assume you have installed this version: [icestudio-0.5.1n201120-win64.exe](https://s3.eu-west-3.amazonaws.com/icestudio.io/nightly/201120/icestudio-0.5.1n201120-win64.exe)

Test #1:  Uninstall python 2.7, execute icestudio and click again on Tool/Toolchain/Install

Let me know what happens

ph1lj-6321, 2020-11-25:
Unfortunately that gave me the same problem - 10% in Error trying to install Virtual Enviro ....

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

[EXTERNAL EMAIL]

This message is intended only for the named recipients. This message may contain information that is confidential or exempt from disclosure under applicable law. Any dissemination or copying of this message by anyone other than a named recipient is strictly prohibited. If you are not a named recipient or an employee or agent responsible for delivering this message to a named recipient, please notify us immediately, and permanently destroy this message and any copies you may have. Warning: Email may not be secure unless properly encrypted.

ph1lj-6321, 2020-11-25:
See the attached - error

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

Obijuan, 2020-11-26:
I cannot see the attached error. Could you please send it again? You can check the issue you opened for checking what developers can see:  https://github.com/FPGAwars/icestudio/issues/462

ph1lj-6321, 2020-11-26:
Ok - here it is
[image]

Obijuan, 2020-11-26:
It seems that there is a missing dependency. The software for downloading the toolchain is not executed. That piece of software is installed in the computers we have tested but it is not in yours, so let's try to find what is missing. Once we detect that, we will update icestudio to show a more human-readable error message with instructions on how to fix the problem...

# Test2: Please, open a command line (window terminal), execute this command and show me the output:

pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio

ph1lj-6321, 2020-11-26:
Here you go

[image]

Obijuan, 2020-11-26:
Great! I can see the problem now. The nightly-builds use the latest toolchains that are downloaded from github. So you need to install git. Just download the installer from here and execute it:

https://git-scm.com/download/win

Once you have installed it, check again this command to make sure that it is working (no red messages on the terminal):

pip3 install -U git+https://github.com/FPGAwars/apio.git@develop#egg=apio

And then run icestudio and install the toolchain

ph1lj-6321, 2020-11-26:
Uhm - after a long time downloading the apio - seemed to install that ok, started the icestudio toolchain install and got original error again

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

[EXTERNAL EMAIL]

https://git-scm.com/download/win

Obijuan, 2020-11-26:
the command pip3 executed on the console was just for knowing what the problem is (it installs apio, but not for icestudio). Git was missing, but there are more missing dependencies or another problem. Try to execute icestudio as an admin, just to try to collect more information on what is going on.

In the meanwhile I am going to switch to windows and do some tests for finding another solution

ph1lj-6321, 2020-11-26:
Here's what I get after running icestudio as admin

[image]

Obijuan, 2020-11-26:
can you  show me the content of the folder c:\Users\100392479\.icestudio please?  I really have no idea what's going wrong, i need to have a look to see if i can find something (i wish i were there)

ph1lj-6321, 2020-11-26:
I know that feeling, by the way I was able to run the standard Win64 bit version, but decided to try the Nightly build . . .

[image]

Obijuan, 2020-11-26:
nightly version is much much better (the stable is quite old)  
Let's try to install the toolchain manually. I need to see the contents of the venv/bin folder please

ph1lj-6321, 2020-11-26:
ven/lib - only

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

[EXTERNAL EMAIL]

Obijuan, 2020-11-26:
ok. It looks like the installation of the virtual environment failed. I will think about it and tell you more tests to do

Obijuan, 2020-11-26:
I need you to open a command line an execute

python

(I wan to see whats the default python that is executed)

and then this command:

python c:\Users\100392479\.cache\virtualenv-16.7.10\virtualenv.py  c:\Users\100392479\.icestudio\venv

ph1lj-6321, 2020-11-26:
This what I get back

[image]

Obijuan, 2020-11-26:
ok, you have python3.9. That's ok

The second error is due to a typo in my previous message. This is the correct one (I hope):

python c:\Users\100392479\cache\virtualenv-16.7.10\virtualenv.py c:\Users\100392479.icestudio\venv

Not being in front of the machine and without copy&paste makes things dificult hahahaah

ph1lj-6321, 2020-11-26:
Probably dont want to this but . . . 
[image]

ph1lj-6321, 2020-11-26:
I dont not have the cache folder

Obijuan, 2020-11-27:
ups.... was my mistake again. The character backlash is special and github remove them. The correct name of the folder is .cache

This should be the command:

```
python c:\Users\100392479\.cache\virtualenv-16.7.10\virtualenv.py c:\Users\100392479\.icestudio\venv
```

ph1lj-6321, 2020-11-27:
Sorry that didn't work either ;-

[image]

Obijuan, 2020-11-27:
ok. Let's install the toolchain in a different way:

1. From the file manager, enter to the `c:\Users\100392479\.icestudio` folder
2. Delete all the folder in there: venv, .cache and profile.json
3. Open the command line an cd to the `c:\Users\100392479\.icestudio` folder

```
cd c:\Users\100392479\.icestudio
```

4. Create a new python virtual env:

```
python -m venv venv 
```

5. Install apio

```
venv\Scripts\pip install apio==0.5.4
```

6. Install apio packages

```
set APIO_HOME_DIR=C:\Users\Obijuan\.icestudio\apio
venv\Scripts\apio install -a
```

If all the commands were ok, execute icestudio

ph1lj-6321, 2020-11-27:
Getting there - got this

[image]

I then ran icestudio and I selected the iceFun package from within icestudio, selected to install the toolchain and got original error

ph1lj-6321, 2020-11-27:
Did the same with the BX module, same error

Obijuan, 2020-11-27:
The toolchain has been already installed (by means of the external commands), so you do not have to install it again. Just select the board and start working.  For whatever the reason, the toolchain installation from icestudio is not working in your computers (that is something that we should investigate further), but icestudio 0.5.1n201120 should work now. Is it?

If it is not working maybe it is because the installation from inside icestudio broke something. If htat is the case, repeat the previous steps, install the toolchain from outside icestudio, run icestudio, select the board and start working

ph1lj-6321, 2020-11-27:
I can not verify or build the design - as it is warning me to install the toolchain - despite having done the manual process again.

Obijuan, 2020-11-27:
I am really confused. I've tested that procedure in two computers (with linux and windows) and It worked. So, at this point, I do not know what is happening and I've run out of ideas. So I will wait for this error to show up in another computer and then getting more information. I feel really frustated for not knowing what's going on

ph1lj-6321, 2020-11-27:
I have seen this mentioned a few times in the community - with recommendations to ensure Python 2.7 is not at the top of Path, I tried those suggestions too.

ph1lj-6321, 2020-11-28:
I have got it to work on another machine, no issues - this has Win 10 Pro Version 1909,  OS build
18363.1198

The org machine (we were working with) has Win 10 Enterprise version 1809, OS build 17763.1039

Not sure if that helps - I full understand the frustrations

Obijuan, 2020-11-29:
Glad to hear that! At least you have it working  
Thanks for the information. We will try to reproduce.  In Icestudio develpment I've changed the way the virtual python machine is created (based on the command I told you to execute on the console). I hope that it fixes the problem. When we release the next nightly I will let you know for testing

Thanks for you help

ph1lj-6321, 2020-11-29:
Please do

Cheers
Phil

[EXTERNAL EMAIL]

ph1lj-6321, 2020-11-29:
One other thing - a fellow faculty member also uses the win 10 staple version, his does work but it takes 10 times longer to build any design (he too has Win 10 Enterprise) - when compared to my Pc which uses, Win 10 Pro - I also noticed the Much long build times - when I had the staple version on the laptop with Enterprise,  that‘S when I decide to try the nightly build.

Relevance - not sure, but just wondering

Cheers
Phil

overtinkering, 2021-04-17:
I tried everything in here and to no avail. I wonder why is the installer forcing me to install python 3.8 when 3.9.4 (and 2.7 as well, and tried changing the path order) is already installed.
Once the install ends succesfully, Trying to run icestudio fails with Tthe application was unable to start correctly (0xc000007b)."

ph1lj-6321, 2021-04-19:
Any ideas on this ? - Anyone !
Love to switch over to the latest build, but needs to work on Win 10 Enterprise

Obijuan, 2021-06-29:
It would be great if you could test icestudio 0.6.0-rc2:

https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0-rc2

We have fixed some bugs related to the installation. I am curious to see if your problem has been fixed or not

Obijuan, 2021-07-05:
The 0.6.0 will be released by July 10th. If no feedback by then, we will considered this issue fixed (as other win10 users has not reported any problem with the instalaltion)
Your feedback and tests are very very very welcome
We are looking forward to hearing from you
Thanks

ph1lj-6321, 2021-07-09:
Hi
Not sure, what is wrong with your email system - but I replied to this - NO it still does not get beyond 10% of the install process, when using it with Win 10 Enterprise.
In addition please note many of the supplied zip *ice files would not unzip.

Sorry but - you should know.

qpeters, 2021-07-13:
@ph1lj-6321 a new stable release was published (https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0). Does it work?

Obijuan, 2021-07-13:
Hi @ph1lj-6321,
This messages are part of the issue 462. You can see all the messages better on this link: https://github.com/FPGAwars/icestudio/issues/462  
As a fresh new version has been release (0.6.0) it is time to test it from scratch and fix this bug once for all. During the next two weeks I will have some free time so I will work on solving this

These are the steps I want you to follow to recap more information about the problem:

* Step 1:  Install the latest stable icestudio 0.6.0 release from here:  [Icestudio Release 0.6.0](https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0)
* Step 2: Click on View/System Info from the menu. It will show a new window with information regarding your computer and used path. Please, it would be great if you could attach a screenshot

After analyzing that screenshot I will ask you to do more tests

Thank you very much for your help. Your feedback is the only way to solve it

ph1lj-6321, 2021-07-13:
Ok – tried it.

1 not all the *.ice files could be extracted.

2 Same issue, on this Win10 Enterprise laptop – see attached

[PJ_emoji_card]

[EXTERNAL EMAIL]

https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0-rc2

ph1lj-6321, 2021-07-13:
OK, tried the install, when attempting to install the toolchain I get an error

[image]

But when I try to install the toolchain, after 10% I get this

[image]

Obijuan, 2021-07-13:
Great! Thanks!  

It seems that the python venv module does not exist on your system. It is usually installed by defaul in python 3.6. But before testing it further let's collect more information on your system. Please, follow these steps:

* Close icestudio
* Remove the file C:\Users\100392479\icestudio.log  (It is a text file located in you home folder)
* Run Icestudio  (A new icestudio.log file will be created)
* Install the toolchain until the error appears again
* Once the error shows up ("No module name venv"), close Icestudio
* Attach the icestudio.log file to this issue

Thanks you very much for your feedback. We are so close to understand what is going on

ph1lj-6321, 2021-07-13:
here you go
[icestudio.log](https://github.com/FPGAwars/icestudio/files/6810763/icestudio.log)

Obijuan, 2021-07-14:
ok, Thanks. The problem is that, for whatever the reason, your python 3.6 is missing the venv module (that is always installed by default). It is not a problem with icestudio but with your system setup. Open a command line interface (cmd) and type the following command

C:\Python36\python.exe --version

This command just executes python3.6. Is for checking that the python executable file is running ok. It should only shows the python version

This is the problematic command (according to the log file it should fail with the error "no module named venv")

C:\Python36\python.exe -m venv test

What this command does is to create a folder called test and inside it the python virtual environment is created

Please execute these two commands and attach the screenshot. Once the error is confirmed, I will propose some possible solutions

ph1lj-6321, 2021-07-14:
This is what I get  :-

[image]

[image]

Obijuan, 2021-07-14:
All right. Your python 3.6 have missing modules

These are the steps i would like you to follow:

* Remove python 3.6 (As it seems to be broken or incomplete, let's change it by a good one)
* I do not want you to install another version of python. Maybe another app needs this python version. Icestudio works ok with python version greather than 3.4. Install python 3.6.0 using this installer:  

https://www.python.org/ftp/python/3.6.0/python-3.6.0-amd64.exe

download the executable file and run it. Make sure you check the option "Add python 3.6 to path". Choose the "install now option" (It would be great if you could provide a screenshot like the one below) 

[image]

* Once installed, open a cmd line and type the following commands:

python --version

python -m venv test

* The first one is for checking the python version (it should be 3.6.0) and the second one is for creating the virtual environment using the venv module, that should be now correctly installed on your system (let's keep our finger crossed)

ph1lj-6321, 2021-07-14:
Ok - did that, but still no venv

[image]

Obijuan, 2021-07-14:
I would like to see the screenshot of the installer (after the python-3.6.0-amd64.exe is executed)  
I have checked that this installer have the venv module, and that it is installed ok. So we need to find out what is wrong with your system setup (it is something not related to icestudio)  

It would be great if you remove first the python 3.6 and make sure that  the C:\python36 folder no longer exist. Then execute the installer (take a screenshot of the first screen) and then install it again

Finally try the `python --version` and `python -m venv test` commands exactly as I've typed (No C:\python36\ prefix)  

We need to collect as much information as possible

ph1lj-6321, 2021-07-14:
Progress maybe

[image]

Obijuan, 2021-07-14:
Yes, it seems that is working ok with python 3.8.10... That is ok for me... but I would like to know what the problem was in your system (to prevent it for appearing again). So before chaging to python 3.8, let's continue with python36

* Remove the C:\python36 folder
* Run the installer you previously downloaded: python-3.6.0-amd64.exe (it is important to use exactly this installer)
* Take a screenshot of the first window that shows up (it is really important that you take that screenshot. I need to see what is going on)

ph1lj-6321, 2021-07-14:
[image]

then

[image]

finally

[image]

ph1lj-6321, 2021-07-14:
I just tried to run the Add Toolchain and get this
[image]

Obijuan, 2021-07-14:
ok. This is another error. Let's solve them one at a time. So, you have installed in your system multiples version of python, some of them are broken. Other aren't. We've detected that the python 3.6.0 installed in C:\python36 was incomplete. So now you have deteled it. One less problem

So, the python3.6 in your system should  be now the correct one. Let's check that (i am not completely sure as it seems you have multiple python installations)

Open a cmd line and type exactly these commands:

python --version

and

python -m venv test

(without any prefix)

What should be executed is the correct python 3.6.0 with the correct venv module. Let's make sure

ph1lj-6321, 2021-07-14:
[image]

here you go

Obijuan, 2021-07-14:
Great! It seem to work, but do not have access rights to the C:\Windows\system32 folder, so please execute this command instead:

`python -m venv C:\Users\100392479\temp`

ph1lj-6321, 2021-07-14:
[image]

OK

ph1lj-6321, 2021-07-14:
Just an update
Due to time constraints, I had to remove the python3.6.0 install - then installed python 3.8.10 and then build 0.6.0 of IceStudio, all seems to fine on this Win10 Enterprise machine - tested with a couple of simple designs using an iceFUN, and then a BX module.

Obijuan, 2021-07-15:
Good news!! Congrats!  😀️

To recap:

* icestudio 0.6.0 works ok now on Win10 Enterprise
* The problem was a broken python 3.6.0 installed on the system (for all users), which lacked the venv module
* The solution was to uninstall this python 3.6.0 completely (removing the C:\Python36 folder) ...
* and installling a fresh new python 3.8.10

To finish this issue I just need to have your latest icestudio.log file. It would be great if you could attach it to this issue. The file  is locate in your home folder: C:\Users\100392479\icestudio.log

Thank you very much for providing so many helpful information

[1 more reply]

closed 2021-07-15 by ph1lj-6321