FPGAwars/icestudio #463: icestudio no funciona con apio 0.6.0

manuelpas, 2020-11-26:
Hola. instalo apio 0.6.0 con pip y al arrancar icestudio dice que no ay toolchain y instala 0.5.6. si no ay apio icestudio instala 0.5.4 y actualiza a 0.5.6. como se puede cambiar para que icestudio funcione con la ultima version de apio?? obijuan dice que es igual pero es muy raro esto. saludos

Obijuan, 2020-11-27:
Icestudio y apio son independientes. Lo que instalas con pip install no es visible en icestudio. Da igual la versión de apio que instales en el sistema, icestudio se bajará la que fue probada en el momento de su empaquetado

Así, icestudio-0.5.1n201120 instala apio 0.5.4 por defecto. Y se actualiza la toolchain instala 0.5.6 de development. Esta versión 0.5.6 de development es justo la que luego se liberó en apio como 0.6.0

En la próxima distrubición de icestudio (nightly) ya se instalará apio 0.6.0 por defecto

Cierro esta issue. Si en icestudio tienes instalado apio 0.5.6, entonces tienes la última vesión

closed 2020-11-27 by Obijuan

manuelpas, 2020-11-27:
gracias!