FPGAwars/icestudio #464: fecha siguiente distrubición de icestudio??

manuelpas, 2020-11-27:
en la proxima distrubición de icestudio se instalara apio 0.6.0 por defecto
cuando es la siguiente?

manuelpas, 2020-12-02:
no ace falta. encontre como actualizar en twitter: https://twitter.com/JManuarrabal/status/1333885431373586432

[image]

closed 2020-12-02 by manuelpas

cavearr, 2020-12-02:
Hola esta imagen que lanzas es de un fork de icestudio, si estás usando esa versión para cualquier duda dirígete a las issues de ese fork ya que es distinto del proyecto oficial.

manuelpas, 2020-12-02:
la pregunta fue del proyecto oficial pero la ultima apio no esta solo en la nightly