FPGAwars/icestudio #466: npm run dist get error message

albertlu2013, 2020-12-02:
Hi Sir : 

 My system is ubuntu 20.04.1 . node -v : v10.17.0 , python 3.8.5

 I run : npm install    npm start is ok .

 But when I run  'npm run dist'  get the error message as follow:

npm run dist

> icestudio@0.5.1-n201120 dist /home/albertlu/icestudio
> grunt dist

Running "checksettings" task

Running "jshint:all" (jshint) task

   app/scripts/services/compiler.js
    679 |          code += '//---- Top entity'
                                              ^ Missing semicolon.
    810 |      code += '# -- Board: '
                                     ^ Missing semicolon.
    811 |      code += common.selectedBoard.name
                                                ^ Missing semicolon.
    812 |      code += '\n\n'
                             ^ Missing semicolon.
    835 |              code += '[' + pin.index + ']" '
                                                      ^ Missing semicolon.
    838 |              let pullmode = common.selectedBoard.pinout.find(x => x.value == value).pullmode;
                                                                                      ^ Expected '===' and instead saw '=='.
    839 |              pullmode = (typeof pullmode == 'undefined') ? 'NONE' : pullmode;
                                                     ^ Expected '===' and instead saw '=='.
    841 |              code += 'PULLMODE=' + pullmode
                                                     ^ Missing semicolon.
    858 |            let pullmode = common.selectedBoard.pinout.find(x => x.value == value).pullmode;
                                                                                    ^ Expected '===' and instead saw '=='.
    859 |            pullmode = (typeof pullmode == 'undefined') ? 'NONE' : pullmode;
                                                   ^ Expected '===' and instead saw '=='.
    861 |            code += 'PULLMODE=' + pullmode
                                                   ^ Missing semicolon.
   app/scripts/services/tools.js
    908 |        extractVirtualenv,
                 ^ 'extractVirtualenv' is not defined.

>> 12 errors in 34 files
Warning: Task "jshint:all" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! code ELIFECYCLE
npm ERR! errno 3
npm ERR! icestudio@0.5.1-n201120 dist: `grunt dist`
npm ERR! Exit status 3
npm ERR! 
npm ERR! Failed at the icestudio@0.5.1-n201120 dist script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/albertlu/.npm/_logs/2020-12-02T07_57_10_917Z-debug.log

I got another error after adding a semicolon to the file compiler.js 

npm run dist

> icestudio@0.5.1-n201120 dist /home/albertlu/icestudio
> grunt dist

Running "checksettings" task

Running "jshint:all" (jshint) task

   app/scripts/services/tools.js
    908 |        extractVirtualenv,
                 ^ 'extractVirtualenv' is not defined.

>> 1 error in 34 files
Warning: Task "jshint:all" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! code ELIFECYCLE
npm ERR! errno 3
npm ERR! icestudio@0.5.1-n201120 dist: `grunt dist`
npm ERR! Exit status 3
npm ERR! 
npm ERR! Failed at the icestudio@0.5.1-n201120 dist script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/albertlu/.npm/_logs/2020-12-02T07_58_24_621Z-debug.log

I remark "extractVirtualenv," in tools.js
I get the error  message 

Running "toolchain" task
cache/toolchain/venv/bin/pip
> Extract virtualenv zip
Fatal error: ENOENT: no such file or directory, open 'app/resources/virtualenv/virtualenv-16.7.10.zip'

How can I solve this problem ？

Thanks

cavearr, 2020-12-02:
Hi!, update your repo (git pull)  and run again!

Tell us if all is ok!

Obijuan, 2020-12-02:
In the current development version we are changing the way the virtualenv is used. In previous releases we shipped the virtualenv-16.7.zip directly on the icestudio installer. But now it should be previously installed (we are no longer using the virtualenv-16.7.zip file)  
It can be installed on linux with this command:
```
sudo apt install python3-venv
```

@cavearr has fixed a bug, so it should work ok

We are looking forward to receiving your feedback for closing this issue

Obijuan, 2020-12-02:
I've fixed another bug related to this issue

albertlu2013, 2020-12-03:
Hi Juan Gonzalez-Gomez

Yes , It work correctly.

Thanks

closed 2020-12-03 by Obijuan