FPGAwars/icestudio #467: Where are the blocks declare? (as an archive)

fer985, 2020-12-02:
Hello, 
I am doing a final project for college, where I am trying to use Icestudio in other programming languages. I mean I want to reprogram my downloaded copy to be able to read an writhe in VHDL code. But I need to know where is the code of the Blocks saved and if I can remake them so will work with the normal interface, I don't know if any of you know where is that at. Really need the help, and thank you to anyone that can help.

PS: I just want to change it right now in my downloaded project so I won't modify anything that is here so there will be no problem with future users that download it

Obijuan, 2020-12-02:
Blocks are in collection. Icestudio includes initially only one collection: the default-collection. You can find it on this link:

https://github.com/FPGAwars/collection-default

All the blocks are located in the Block folder. The .ice files are just json files (text files) with all the information inside

From icestudio you can directly view the verilog code just by doble clicking on the block itself. So, for example, if you want to see what's inside the Bit 0 block, doble click on this and you will see this:

[image]

All that information is located on this file:

https://github.com/FPGAwars/collection-default/blob/master/blocks/Bit/0.ice

You will see the the property code, with the verilog code for the bit 0:

 "code": "// Bit 0\n\nassign v = 1'b0;"

Also, if you are using the latest nightly 0.5.1n201120, everytime to build a project it will create the folder ice-build which contains the final verilog code of the whole project

manuelpas, 2020-12-02:
#382

closed 2022-01-17 by Obijuan