FPGAwars/icestudio #469: verify Build Upload not work when it launch from' npm start'  or AppImage from ' npm dist'

albertlu2013, 2020-12-03:
Hi Sir:
  I can finished "npm run dist " correctly.
 There are some issues in icestudio : 
1)
  I launch icestudio from 'npm start'.
  These function " Verify , Build , Upload " do not run. The icestudio will re-launch  when I click Verify or Build or Upload. 

2)
I launch icestudio from '/dist/icestudio-0.5.1n201120-linux64.AppIma'.
 The icestudio will NOT re-launch when I click Verify or Build or Upload.
But these function " Verify , Build , Upload " do not run. It never show the "Done".
 
3)
I launch icestudio from '/dist/icestudio/linux64/icestudio'.
The icestudio work fine. I can use Verify, Build,Upload correctly function.

I try it in my ubuntu2004 and linux Mint1804 have the same issues.

How can I solve this ？ 

Thanks !

cavearr, 2020-12-23:
Hi albertlu2013!

Step 1 is fixed yet, try and tell us.

2) I'm needed more information , have you toolchain, drivers... installed?

thanks!

albertlu2013, 2020-12-24:
>    1. I'm needed more information , have you toolchain, drivers...

Hi Carlos

 Now , I try the " npm start " , It is GOOD for everything.
" npm run dist " is successful finished.
But whein I run icestudio in dist\linux64 OR run
icestudio-0.5.1n201120-linux64.AppImage . It can not show any menu bar's
word .
Then I press F12, find 2 errors :
Failed to load resource: net:: ERR_FILE_NOT_FOUND
 IceCollection.jd:1
Failed to load resource: net:: ERR_FILE_NOT_FOUND
 IceHD.js:1

Thanks !

cavearr, 2020-12-24:
Fixed! try it and tell us! i'm refactoring some modules and for this  in develop branch the dist task would be broken sometimes in the next days.

Thanks for testing and report!

albertlu2013, 2020-12-25:
Hi Carlos

YES ,  These are GOOD in  "npm start" , " ./icestudio " or "
icestudio-0.5.1n201120-linux64.AppImage".
Thanks for your fixed it.

closed 2020-12-25 by cavearr

albertlu2013, 2020-12-30:
Hi Carlos:

Today I get the same error when I finish "npm run dist" :
I launch icestudio from '/dist/icestudio-0.5.1n201120-linux64.AppIma'.
These function " Verify , Build , Upload " do not run. It never show the "Done".

I find the error from press F12:
vendors.js:1 Unable to infer path to ace from script src, use ace.config.set('basePath', 'path') to enable dynamic loading of modes and themes or with webpack use ace/webpack-resolver
f @ vendors.js:1
mode-markdown.js:1 Failed to load resource: net::ERR_FILE_NOT_FOUND

Thanks !

cavearr, 2020-12-30:
> Enviados: Miércoles, 30 de Diciembre 2020 11:20:33

Hi! could you send me a .ice file to test with your same or similar file?

albertlu2013, 2020-12-30:
I use the Examples\1BAsic\ONE LED Blink