FPGAwars/icestudio #470: Just test this with both FunICE and Bx

ph1lj-6321, 2020-12-10:
Use apio 0.5.6, I get the error as seen in the image - if I use 0.4.1 however on the the BX at least all is good
[image]

fuesika, 2021-04-19:
I am not able to reproduce your error (with the latest Apio 0.6). Please provide more details or close this issue.

ph1lj-6321, 2021-04-19:
I will try it with Apio 0.6 - will let you know

ph1lj-6321, 2021-04-21:
Uhm - how do you force it use Apio 0.6.**. I have downloaded and installed through pip - it always default to the older Apio
Any ideas ?

fuesika, 2021-04-21:
Did you download Apio through pip? In that case you may need to check with your repository. You can force specific versions by appending "==0.6", e.g. `pip install apio==0.6`

Alternatively, you can try the icestudio AppImage which comes readily packaged with Apio. Downloads are available here: https://github.com/FPGAwars/icestudio/releases

ph1lj-6321, 2021-04-21:
OK - I'm trying to install the Toolchain now using the 5.1w210224 for win64 build - started OK - seems to get stuck around the 50% mark - showing that it's 'installing default apio...' 

It's been an hour - still there.

I'm using a laptop with Win10 Pro

I guess the Win10 install issue hasn't been resolved

ph1lj-6321, 2021-04-21:
Ok - big step forward, with the above process 5.1w210224 was installed with apio 0.6.0 - Thank you.

However the original issue of not building the simple SR example, with the BX is still there - 
[image]

ph1lj-6321, 2021-04-21:
By the way - trued to use the error log, but after the above, my error log folder is empty (yes I did set Preference's Logging enabled.)

ph1lj-6321, 2021-04-21:
Also - I just tried the supplied S-R latch (Examples. 3.Gate 05. SR latch) - same thing with both the BX and IceFUN boards

Obijuan, 2021-04-22:
That particilar example (the S-R latch) does not work by default with the new toolchain (it is not a problem related to IceFun board). It worked ok with the old arachne-pnr tool, but when it was upgraded to nextpnr it stoped working. This new tool, by default, does not allow to have combinational loop. Of course, this can be overrided using the --force or --ignore-loops flags when invoking nextpnr:

https://github.com/YosysHQ/nextpnr/issues/224

But this option has not yet been incluided in apio/icestudio

The S-R latch example should be removed or at least we should provide comments explaining that it does not work with the current toolchain (the --ignore-loops flag should be used)

Obijuan, 2021-04-22:
Insted of impleting the latches and flip-flops directly using logic gates it is better to implement them  in verilog or using the provided blocks. The latest collection related with FF can be found here:

https://github.com/FPGAwars/iceFF

There is no a release yet, but you can download as zip and install it easily

https://github.com/FPGAwars/iceFF/archive/refs/heads/master.zip

It is a work in progress. We are adding new FFs as we need them

cavearr, 2021-04-22:
> By the way - tried to use the error log, but after the above, my error log folder is empty (yes I did set Preference's Logging enabled.)

Hi!,thanks for your feedback, Has you specify in the *Preferences->Logging file* the output file? Are you sure that Icestudio can write at this location?