FPGAwars/icestudio #473: Windows8 UI issue

John-Westlake, 2020-12-21:
Just installed icestudio 0.5n201120 (looking forward to my first steps into the world of FPGA's).

I'm using Windows 8, with the latest Python install (thats installed with icestudio-0.5n201120) 

I'm experiencing UI issues where the text input field "box" for Input and outputs is left expanded (sometimes) after entering the pin I/O (e.g. LED_G)...

See attached image:-

[image]

Thank you for the great software.

John-Westlake, 2020-12-23:
OK, I can conform this issue remains with Python 3.8.7

[image]

John-Westlake, 2020-12-23:
Just created an 32b output box - note how all output fields entry boxes remain expanded:

[image]

cavearr, 2020-12-23:
Hi John! this issue is for the screen resolution, it is in relationship with issue #474 

I'm working on it and tell you soon.

Thanks!

John-Westlake, 2020-12-30:
Cavearr,

A Merry Christmas to you and lets all wish for a better 2021!!!

WRT the "Dropdown" I/O port assignment issue - I'm also encountering that when I mange to correctly assign the I/O pins (I find the restarting the software will open without the "Dropdown" boxes expanded - the I/O ports will RANDOMLY reassign themselves (I'm not sure when this occurs) - so they are initially set correctly, but then when you next look the software has corrupted the port numbers (the software changes the port numbering).

In the case of the attached images:-

Port 5B becomes 25B
Port 2A becomes 22A
Port 3B becomes 13B
port 6A becomes 16A

Correct port  assignment:- 

[image]

Corrupted (randomly by iCEStudio)  port assignment:-

[image]

Also, the software allows duplicate I/O port assignment (and no verification error) - is this correct? 

I'm new to FPGA so there might be a valid reason for duplicate port assignment (how does this work? if port duplication assignment is allowed, are they logically OR'ed)? but for me it be sure useful to have an Option where I/O ports cannot be assigned twice (atleast outputs)...

cavearr, 2021-01-04:
Hi John! i'm working on it is clearly a bug assignment, thanks for your tests and feedback! i'll try to fix soon.

John-Westlake, 2021-01-07:
Hi cavearr,

So I've encountered more I/O port define issues (at least on the iCE40 UltraPlus Breakout Board).

I'm getting a pink "Duplicated pins" error although pins dont appear to be duplicated (unless there is some form of limitations I'm unaware of on the UltraPlus Breakout Board).

Moving the I/O pins to a different ports (I/O bank) resolves the issue, but I cannot use all I/O pins - the error seems unpredictable.

Duplicate pin error (note the O5 to O0 I/O port assignment) on the far right hand side:

[image]

No "duplicate port" error message when rearranging pins:-

[image]

John-Westlake, 2021-01-07:
I've also encountered possibly odd behavior (unless I dont understand the purpose of the "FPGA pin" set function).

When I "Add a block"  - the internal input / output "connection" boxes are reconfigured to "FPGA pin" although they where not set when the block was created - as a block the input / output connections should just be an internal connection (NOT an FPGA external I/O pin).

[image]

When the same block is opened with "File open", then the internal input / output "connection" boxes remain in there original (correct / intended / wanted) state.

[image]

I think the desirable operation is NOT to change the "FPGA pin" states of the I/O connections when importing a Block unless the designer manually reconfigures them.

John-Westlake, 2021-03-26:
Issue still not fixed on the latest build 0.5.1w210224 :(

cavearr, 2025-01-08:
fixed in last wips https://downloads.icestudio.io

closed 2025-01-08 by cavearr