FPGAwars/icestudio #474: Win8 UltraWide Monitor limited working area

John-Westlake, 2020-12-23:
Win8 / icestudio 0.5n201120 / Python 3.8.7

I'm using an Ultra Wide Monitor (5120 x 1440, driven via x2 HDMI ports so the OS sees x2 2560 x 1440 "monitors" side by side) , and there seems to be an IceStudio working area limit of around 3000 pixels wide (I cannot tell the exact pixel limit, but its greater then 2560) - the remaining monitor area is "White" apart from the toolbars.

[image]

Can the "working area" limits of IceStudio be increased so we can make the full use of UltraWIde monitors which are becoming more common.

John-Westlake, 2020-12-23:
In the attached image the White area is unusable for the logic layout, some icons are displayed on the far right.

[image]

cavearr, 2020-12-23:
Hi John! i doesnt test icestudio in this kind of monitor! thanks for your probes!

I'm trying to patch it as soon as possible.

I'm maintaining open this wire to update the progress.

Thanks!

cavearr, 2021-01-07:
Hi John! you can download a new nightly from https://downloads.icestudio.io (0.5.1n210107)

In this nightly your problem should be fixed.

¡Tell us!

John-Westlake, 2021-01-07:
Hi Cavearr,

Brilliant , thank you - I can confirm that after a quick test the issue is resolved at 5120 x 1440, I also tested with 5120 x 2880 and also works well - so I hope this now resolves the issue for all :)

Once again thank you for your brilliant work!

cavearr, 2021-01-07:
Thanks to you! for use icestudio and for your feedback!

closed 2021-01-07 by cavearr