FPGAwars/icestudio #475: MacOSX Nightly Toolchain installation stalls in `apio installs icesprog`

undo76, 2020-12-29:
I'm trying to install the latest Nightly version but it fails while installing the toolchain. 
I have tried to reinstall it from scratch several times with no success. The stable version is working properly.

[image]

As I understand the IceSugar support is not available in MacOSX, so maybe it shouldn't try to even download it at all.

Obijuan, 2020-12-30:
There is a bug in icestudio 201120 that prevents it from installing the latest toolchain. It has been already fixed in the development version but the new binaries has not already been created

As a workaround, I've make a quick patch to apio: the version number is lowered to 0.5.99 and now it should installs ok

Please update the toolchain again (from your icestudio 201120). After the upgrade, you should see apio 0.5.99 in Tools/Toolchain

Waiting for your new feedback. Thanks for your help

undo76, 2020-12-30:
The workaround is working properly. Thanks a lot for your prompt response!

Obijuan, 2020-12-30:
Great! 😀️ Thanks for the Feedback

closed 2020-12-30 by Obijuan