FPGAwars/icestudio #476: Probably In FAQ:  How To Add Custom Board?

whathisface, 2021-01-04:
Hello,

I'm sure this has been asked before - I see there's support for Lattice's iCE405K devices - can custom PCF files be made and 
designs be done using that as a starting point?

Would be nice to build the FPGA files based on custom PCF's and possibly other directives - maybe this has been done already?

I saw one open issue for documentation but I looked at the docs - maybe I missed this?

Great idea for this and I just looked at the tool under Win10 and generated some simple Verilog - nice!

Like the suggestion said - part/pins/pcf and then example Verilog can be generated?

Thanks In Advance,
whf

h4m3ds, 2021-03-07:
My problem too, I just added iceZero board and it can be selected as board and verified. But build fails and complains "Board not found"

John-Westlake, 2021-11-20:
Also, it be great if iCEStudio could support the iCE40 FPGA family without "Dev Boards" - so the iCE40 family & package type (BG121 / CB132 / CT256 etc) could be selected much the same way as the current "Boards" are selected, but with the option to select a programmer type and target FPGA clock frequency.

The Lattice iCEProg programmer (as used on the Lattice iCE40-HX8K Breakout Board etc.) seems to be one of the best options to program "Custom" FPGA boards / designs....