FPGAwars/icestudio #479: Icestudio nightly 0.5.1n210107

Democrito, 2021-01-07:
Hola Carlos, te comento lo que ha sucedido en Windows 10 home:

Una vez instalado la nueva nightly al hacer doble clic sobre un ICE no se abría. Reinicié el PC y al volver a intentarlo seguía sin abrirse. Se creaba un archivo llamado "debug.log" y en su interior contiene este mensaje:
<<
0107/065736.680:ERROR:http_transport_win.cc(175)] WinHttpCrackUrl: La operación se completó correctamente. (0x0)
[0107/065847.383:ERROR:http_transport_win.cc(175)] WinHttpCrackUrl: La operación se completó correctamente. (0x0)
[0107/065910.922:ERROR:http_transport_win.cc(175)] WinHttpCrackUrl: La operación se completó correctamente. (0x0)
[0107/070336.493:ERROR:http_transport_win.cc(175)] WinHttpCrackUrl: La operación se completó correctamente. (0x0)
>>

Reinstalé Icestudio, entonces me salió un error de una DLL faltante, el mensaje decía que si volvía a reinstalar resolvería el problema (es algo que a veces sucede en Windows). Al volver a instalar Icestudio (tercera intento) ya funciona como se espera.

Por lo demás, al menos hasta el momento todo parece correcto.

Saludos y gracias!

cavearr, 2021-01-07:
Muchas gracias Demócrito! como siempre así da gusto.

Tenemos que revisar el tema de windows cuando cambiamos de versión, porque se ve que el nwjs deja algo por ahí perdido y no le gusta hasta que no se reinstala o se pisa, lo investigaré.

Dejo la issue abierta para ir llevando el seguimiento.

Gracias!

cavearr, 2021-01-07:
HAciendo una primera búsqueda parece ser que es un problema de nwjs con las nuevas versiones de windows, en sí no tiene nada que ver con Icestudio.

Lo malo es que el parche lo han hecho para versiones qeu ahora mismo no podemos actualizar de nwjs (estamos estancados por unas dependencias de core importantes ).

Aquí la referencia del problema:

https://github.com/nwjs/nw.js/issues/7450

Parece que sólo hay que "reintentar" y acaba funcionando, puede ser algo molesto si ocurre pero de momento es lo que tenemos.

Volveré a darle una vuelta a ver si podemos salir del atasco de la versión de nwjs en la que estamos que la verdad estaría bastante bien.

Gracias por detectar esto!

Democrito, 2021-01-07:
Para mí no hay molestia alguna, es mecánica de funcionamiento, porque funciona. Hemos de pensar que en proyectos de esta envergadura es una lucha constante con cosas que no depende de nosotros.

Un abrazo!

cavearr, 2021-06-08:
Cierro esta issue ya que en las nuevas wip parece solucionado, si vuelve a aparecer la reabrimos.

closed 2021-06-08 by cavearr