FPGAwars/icestudio #480: Icestudio nightly 0.5.1n210107 large code block Critical UI rendering issue Win8

John-Westlake, 2021-01-15:
Icestudio nightly 0.5.1n210107 large code block UI rendering issue Win8 / Python 3.8.7 (2560 x1440 screen resolution).

I've encountered a repeatable UI rendering issue when "Zooming into" longer Code blocks:

1. Long code block - say 70 lines

2. Zoom into code block "Window" so you enlarge the text and can read / edit the code (so your viewing only a portion of the code)

3. Scroll to the bottom lines of the code - viewing code below current the Zoomed visible screen area (so the visible text "Area" has to scroll down the screen)

4. Scroll back up the screen and the Lower iCEStudio taskbar is now "stuck / attached" to your code block - moves up and down the working UI Window as you Scroll trough the code...

Images attached:

[image]

[image]

Also, any progress please on issues #473 ?  "Win8 UI issue & I/O Port define corruption also odd "FPGA pin" set behavior
#473" its causing constant issues....

John-Westlake, 2021-03-26:
Issue still not fixed on the latest build 0.5.1w210224 :(

cavearr, 2025-01-19:
This is fixed in the last wip https://downloads.icestudio.io if you need or detect some other error,please reopen

closed 2025-01-19 by cavearr