FPGAwars/icestudio #481: 0.5.1n210107 REALLY Critical issue, iCEStudio Verifys / Builds / Saves earlier edited code / design then displayed

John-Westlake, 2021-01-15:
Win8 / Icestudio nightly 0.5.1n210107 / Python 3.8.7 (2560 x1440 screen resolution).

A really odd issues that keeps reoccurring:

iCEStudio Verifys / Builds / **AND Saves** earlier edited "code / design" then is currently being displayed and edited.

First clues to the issue are code complies with errors, when there is no apparent error in the code. RED verification error "crosses" can be displayed next to empty code lines. (I'm guessing they align with earlier code - but not the current version of code being edited / displayed).

When in this apparent "corrupt" state, Saving the code saves an earlier version of the edited code - so your **CURRENT work is LOST**... if in doubt "copy and past" the code to a new session of iceStudio.

At first I could not understand what was wrong with my code, the verification pointed to errors on empty lines, and then saved earlier versions of the edited code - something very odd is going on...

MAYBE A CLUE... (not sure if this is what happens - but sometimes I work like this):

I open a file, say "I2S input Stage A01aa" in a first session of iCEStudio; then with this file still open - I open a second session of Icestudio with the same file "I2S input Stage A01aa" - but in the second iCEStudio session  I rename and save the file as "I2S input Stage A01aaa" before I start working with the second iCEStudio session - (with the first session of iCEstudio still open as a "code reference" ).

Its possible its seems that iCEstudio is still linked somehow to the first opened session / instance of "I2S input Stage A01aa"  although I'm working on second open session - although it has been renamed and saved as "I2S input Stage A01aaa".

I've also noticed such odd behavior when I "Copy and Paste" a "Block" in one opened session of iCEStudio and work with it in a second concurrently opened session of iCEStudio (modifying the Block etc) - although I save the file in the second session, its somehow still linked to the first opened instance of iCEstudio - and modifications etcs are lost when saved, with existent verification errors with the second sessions modified code - although there are apparently non (it seems somehow that the verification / build still seems to be somehow "loosely" linked to the original code in the first still opened session).

What you see in the current edit / working windows is not the version that is verified / built / saved!!!

Its Hard to explain - but non existent verification errors (even indicating errors on empty lines),  lost modifications / edits to "blocks" and Files saved with EARLIER code is occurring all the time... its frustrating as not only do you loose your work when "missed saved", but also spending much time searching for non existent errors - making it even more confusing to a beginner who is not confident in his code abilities...

I understand the risks of Nightly codes, but the official build is years old and missing all the past years updates and improvements :)

Bug report - not complaints - thank you for such great software, allowing the self thought and too old to learn "non skilled" such as myself to work with FPGAs :)

cavearr, 2021-01-15:
Hi John!! thanks a lot for your bug report!!!!, could you attach the .ice to try us to reproduce your case?

Thanks!

John-Westlake, 2021-01-15:
So this is an earlier version of the code (with errors) that was being worked upon, the updated version was not correctly saved and has been lost...

[I2S input stage a01aaa.zip](https://github.com/FPGAwars/icestudio/files/5819474/I2S.input.stage.a01aaa.zip)

I've since rewritten the "missed saved file" (still work in progress - with errors, I was working on the errors when I encountered the issues) - this is close to what the missing "missed saved" version would have looked like:

[I2S input stage a01aaaa.zip](https://github.com/FPGAwars/icestudio/files/5819503/I2S.input.stage.a01aaaa.zip)

When working on the earlier version that "did not save the updates" the verification / build was pointing to errors that did not exist - the "same" code works OK when recreated again (to replace the lost modifications)....

cavearr, 2021-01-15:
Hi John!! don' worry for your code, the important is learn every day, practice and enjoy! Each day you grow and improve your skills!!

thanks a lot for your .ice, i'm playing with it to try to reproce your error.

Tell you soon!

John-Westlake, 2021-01-16:
Cavearr,

Thank you for your support and for such powerful and useful software - without it I could never have got this far in such a short time.

It be great to meet oneday to discuss improvements / extra features (and just to meet)!!... if you ever find yourself in the Czech Rep. Ping me!

John-Westlake, 2021-01-16:
Carlos,

I know trying to find this issue is going to be hard - I manged to encounter the issue again just now when updating a "Block".

1. Two iCEstudio session open.

2. I updated a "Block" design (originally called "BCLK Ratio detector") -  I updated the design by adding an "fs_err" pin and extra "verilog" code to the "fs detect cnt" to drive this pin. 

I renamed and saved this new block "BCLK Ratio detector **with clk_error A01a**"

Image and code below:-

[image]

[BCLK Ratio detector with clk_err A01a.zip](https://github.com/FPGAwars/icestudio/files/5823893/BCLK.Ratio.detector.with.clk_err.A01a.zip)

The second concurrent session of iCEStudio is also open and has a test design with the **original unmodified Block loaded** the Block called "BCLK Ratio detector".

I update the "BCLK Ratio detector" with the updated design "BCLK Ratio detector **with clk_error A01a**"

Everything appears correct on the screen (note the **extra fs_err pin** on the design):

[image]

[BCLK Ratio detector with clk_err A01a test.zip](https://github.com/FPGAwars/icestudio/files/5823906/BCLK.Ratio.detector.with.clk_err.A01a.test.zip)

But the design is now corrupted - it fails to build as the new updated Block "BCLK Ratio detector **with clk_error A01a**" still holds elements of the earlier version block "BCLK Ratio detector". **NOTE** that the "fs detect cnt" has **NOT** been updated with the latest version of the Verilog code with the extra fs_err pin:

[image]

It seems that iCEStudio has does not update a "circuit block" in this case (fs detect cnt), when its been opened with an earlier instance of the design...

While this an an example elements of a corrupted "Block"  - I've had the same issue with WHOLE designs - whats seen on the screen is not what is saved / built. Some elements are updated, others not...

Sorry, I hope you can understand alittle my explanation... its very hard to explain the process that results in the errors.

Once again, please forgive my code, the test example above uses some of my very first attempts at Verilog - as you say each day I hope I've been getting better!

John-Westlake, 2021-03-26:
Issue still not fixed on the latest build 0.5.1w210224 :(

SiccoDwars, 2021-06-08:
On "0.5.1n210107 REALLY Critical issue, iCEStudio Verifys / Builds / Saves an earlier edited design then currently displayed
#481" - the same thing happened to me also. Using a PC Windows10, 64bit. It seems to relate to another annoying bug: using UNDO (ctrl-Z) when editing code in a code block. After having touched ctrl-Z, you're in trouble. Only way out is to quit Icestudio and restart. And without regular save's all edits before accidentally touching Ctrl-Z are gone. Have updated to many overnight builds, but still this issue strikes. Now at 0.5.1w202106040506 where it just happened.  

UNDO by Ctrl-Z does work as expected when in visual mode working with blocks. But in text mode when editing Code, the problem strikes. Normal cut and paste with Ctrl-X, Ctrl-C, Ctrl-V does work ok inside Verilog code text. Undo Ctrl-Z does not and has pretty dramatic consequences. Which don't show immediately. But does begave like disconnect between what's on screen and will not get saved and cannot be recovered versus what's the "earlier edited design than currently displayed"

qpeters, 2021-07-12:
@John-Westlake @SiccoDwars is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

SiccoDwars, 2021-07-13:
Just tried in v0.6.0, win64. Not solved it seems. Here’s how I test, and how I assume others can reproduce the issue: Open Icestudio, insert a Code block with only one output x, and start editing. I typed in test 123, enter, enter, space, and then ctrl-Z (which did not undo typing my last space, but that’s sort of tolerable), and then test 12345 on a next line. Then File, Save to save. That’s the screenshot below. Then File, New, and Open what was saved. I then get the next screen dump, which shows that only my edits prior to touching ctrl-Z are saved, everything after that ‘incident’ went into a black hole. Despite the first screen clearly suggesting in its green popup that all was properly saved when saving...    Sicco ***@***.***+31 629421058   From: qpetersSent: 13 July 2021 01:33To: FPGAwars/icestudioCc: SiccoDwars; MentionSubject: Re: [FPGAwars/icestudio] 0.5.1n210107 REALLY Critical issue, iCEStudio Verifys / Builds / Saves an earlier edited design then currently displayed (#481) @John-Westlake @SiccoDwars is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

SiccoDwars, 2021-07-13:
Just tried in v0.6.0, win64. Not solved it seems.

Here’s how I test, and how I assume others can reproduce the issue:

1)	Open Icestudio, insert a Code block with only one output x, and start editing. I typed in test 123, enter, enter, space, and then ctrl-Z (which did not undo typing my last space, but that’s sort of tolerable), and then test 12345 on a next line. 
2)	Then File, Save to save. That’s the screenshot below. Then File, New, and Open what was saved. I then get the next screen dump, which shows that only my edits prior to touching ctrl-Z are saved, everything after that ‘incident’ went into a black hole. Despite the first screen clearly suggesting in its green popup that all was properly saved when saving... 
[image]
[image]

cavearr, 2021-07-13:
Hi! i'm working now on it, tell you soon

Obijuan, 2021-07-13:
The 0.6.0 stable release does not fix this issue, that is still pending  
On the 0.6 we have only focused on the toolchain installation  
The development work continue in the develop branch (wip versions). We hope we can solve it as soon as possible, but we are very few developers trying our best in our limited time

cavearr, 2021-07-13:
Hi SiccoDWars!

Could you try the latest wip 0.6.1w202107130607 ? and tell us if all works ok?

https://downloads.icestudio.io/

Thanks!

SiccoDwars, 2021-07-13:
Hi, Yes, the issue seems fixed now in 0.6.1w_today. You’re over asking a bit when asking me if ALL works ok. But the ctrl-z bug is fixed! Many thanks to you and team. While we’re talking, here some suggestions for more improvements possibly: A button to cancel an ongoing Build or Upload. Quite often I realize seconds after pressing ‘Upload’ that I made a typo or coding error. But then I have to wait until the ongoing action is completed. That can be minutes. Being able to cancel and go straight back into editing would be much appreciated. The Verify/Build/Upload pushed to the background with a local copy of the relevant source files maybe? Just so that the editor isn’t blocked while compilation/synthesis/programming is in progress.A ‘replace block’ option. Often I have a Code block with many inputs and many outputs an parameters that got a tweak. When I do such a  ‘block replace’ action now, I delete the old block and then do ‘File’, ‘Add as block..’, But then I need to rewire all inputs and outputs and parameters again. With output wires often going to multiple other blocks… So quite likely I miss a line or two when rewiring… A way to let a WIDTH (# bits) local parameter trickle into a Code block so that array input and output busses automatically get the widths that they need. For example I have generic Verilog code for a N bits adder block. I want a generic pre-cooked Collection with a 4 bits, 8 bits, 12 bits, 16 bits, 24 bits and a 32 bits version of that binary adder. Verilog as such allows for that flexibility. As in one and the same Verilog module can do all those different array WIDTHs. But if I make Icestudio blocks, there’s a lot of editing to do as I need to edit each [array] input and output width, both in the specification dialog ‘Enter the input ports, Enter the output ports’, and then again in the Input and Output spec when wrapping such a Code Block into a .ice file for Importing as a Block later on.  Regards, Sicco ***@***.***+31 629421058   From: Carlos Venegas ArrabéSent: 13 July 2021 20:51To: FPGAwars/icestudioCc: SiccoDwars; MentionSubject: Re: [FPGAwars/icestudio] 0.5.1n210107 REALLY Critical issue, iCEStudio Verifys / Builds / Saves an earlier edited design then currently displayed (#481) Hi SiccoDWars!Could you try the latest wip 0.6.1w202107130607 ? and tell us if all works ok?https://downloads.icestudio.io/Thanks!

cavearr, 2021-07-13:
Hi! thanks for your feedback!! i'm close this issue ok? but i read your suggestionns and add as independent issues because there is very interesting things in your comments (there is some features that are work in progress now! ;) )

Thanks a lot!

closed 2021-07-13 by cavearr