FPGAwars/icestudio #484: Doesn't work for me with Fomu

fancellu, 2021-01-27:
I go to upload, I get this
```
dfu-util -d 1209:5bf0 -D hardware.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 1209:5bf0
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 4096
Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Error: File ID 0000:0000 does not match device (1209:5bf0 or 1209:5bf0)
scons: *** [upload] Error 74
```

dfu-util -l shows
```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2019 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Cannot open DFU device 2972:0047
Cannot open DFU device 2972:0047
Found DFU: [1209:5bf0] ver=0101, devnum=3, cfg=1, intf=0, path="2-1.4", alt=0, name="Fomu PVT running DFU Bootloader v2.0.3", serial="UNKNOWN"
```

Ring a bell for anyone?

icestudio works fine for me with my icebreaker

cavearr, 2021-03-10:
What version of icestudio do you have?

fancellu, 2021-03-10:
v0.5.1n210107

However just tried 0.5.1w210224, same result

```
set APIO_HOME_DIR=C:\Users\Dino\.icestudio\apio& "C:\Users\Dino\.icestudio\venv\Scripts\apio.exe" upload --board fomu --verbose-pnr -p "e:\Program Files\icestudio\resources\collection\examples\1. Basic\ice-build\01. One LED"

[Wed Mar 10 07:07:31 2021] Processing fomu
--------------------------------------------------------------------------------
dfu-util -d 1209:5bf0 -D hardware.bin
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 1209:5bf0
Run-time device DFU version 0101
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0101
Device returned transfer size 4096
Warning: Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Error: File ID 0000:0000 does not match device (1209:5bf0 or 1209:5bf0)
scons: *** [upload] Error 74
========================= [ ERROR ] Took 0.57 seconds =========================
```

qpeters, 2021-07-12:
@fancellu is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

Obijuan, 2022-01-17:
It should be working ok in the latest version (0.8.2w). Feel free to open a new issue if the error persists

closed 2022-01-17 by Obijuan