FPGAwars/icestudio #485: Add Digilent  Nexys A7 Artix-7 FPGA

veso266, 2021-02-13:
Hi there, not sure how easy is to add another board to this program, normaly you have to use [Vivado](https://www.xilinx.com/products/design-tools/vivado.html)

but I wanted to try Icestudio it seems easy to work with and it better beginner frendly then Vivado is

Its boardfiles are here: https://github.com/Digilent/vivado-boards/archive/master.zip if it helps any

qpeters, 2021-07-12:
SymbiFlow is an alternative to Vivado: https://symbiflow-examples.readthedocs.io/en/latest/ (https://github.com/SymbiFlow/symbiflow-examples)

AcThPaU, 2022-10-11:
The Symbiflow tool chain is completely stable now: https://f4pga.org/
I can't see why it cannot be added to icestudio.

veso266, 2023-06-05:
Oh thats great
why does it need Symbiflow tool chain, I thought IceStudio uses APIO, but if I saw this APIO does not support  Nexys A7 Artix-7 FPGA

cavearr, 2023-06-06:
Hi! we are working on it in the next months this feature will be available, maintain this thread open to up to date.