FPGAwars/icestudio #486: Support for upduino 2.1 broken.

wgaylord, 2021-03-10:
Not sure if this is an icestudio issue or an apio issue but once again icestudio outputs the following error while trying to build a simple button example.

```
set APIO_HOME_DIR=C:\Users\chibi\.icestudio\apio& "C:\Users\chibi\.icestudio\venv\Scripts\apio.exe" build --board upduino21 -p "C:\Users\chibi\AppData\Local\Temp\icestudio-9744TPtyBDnZfy9G"

[Wed Mar 10 01:46:41 2021] Processing upduino21
--------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q main.v
nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf main.pcf -q
ERROR: package does not have a pin named '0' (on line 3)
ERROR: Loading PCF failed.
0 warnings, 2 errors
scons: *** [hardware.asc] Error -1
========================= [ ERROR ] Took 1.64 seconds =========================
```

My Icestudio version is 0.51n210107 the latest nightly build off the site.
The APIO version according to ice studio is 0.6.0

wgaylord, 2021-03-21:
Anyone have a fix for this? Or know whats causing it?

cavearr, 2021-03-30:
hi! Could you try the last wip version?:

https://downloads.icestudio.io/

Uninstall icestudio and install the new one.   Then update apio from Icestudio and try it.

If error continues, tell us.

Thanks!

wgaylord, 2021-03-31:
So the latest wip has been stuck at "Installing default apio" after I set the board for over an hour. So not making much progress.

wgaylord, 2021-04-02:
After letting it sit over night it was still stuck on the Installing default apio stage of setting up the board's toolchain initially. Any ideas?

Obijuan, 2021-04-09:
Please, try to install the toolchain manually by following these instructions:

https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio

cavearr, 2021-04-09:
Hi!, review this howto :

#494

and attach the logging file.

wgaylord, 2021-04-09:
I just tried the zip version of the WIP and that worked, not sure why the installed one wouldn't install apio or why the normal build stopped working either.

qpeters, 2021-07-12:
@wgaylord is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

Obijuan, 2022-01-17:
Very likely, the initial problem was caused because the pin was not assigned  
Anyway, the upduino 2.1 seem to work ok on the latest icestudio (0.8.2w)  
I will close this issue. Feel free to reopen it if the error persists

closed 2022-01-17 by Obijuan