FPGAwars/icestudio #490: icestudio 0.5.1n210107 linux64 nighty

sasasuio, 2021-03-30:
Hola,

A ejecutar la versión nighty en linuxmint 64 bits 0.5.1n210107 arroja el siguiente error en segundo plano al tratar de instalar el toolchain:

command failed: /usr/bin/python3.8 -m venv "home/username/.icestudio/venv"

luego queda en 10%, hice un rollback a la anterior y funciona bien :) 

Espero sirva de ayuda,

Obijuan, 2021-03-30:
Tienes que instalar el entorno virtual de python. Si tu distribución de Linux es Ubuntu, ejecuta este comando desde la línea de comandos:

sudo apt install python3-venv

Luego arranca la última versión de Icestudio

sasasuio, 2021-04-04:
> Tienes que instalar el entorno virtual de python. Si tu distribución de Linux es Ubuntu, ejecuta este comando desde la línea de comandos:
> 
> sudo apt install python3-venv
> 
> Luego arranca la última versión de Icestudio

Funciona perfectamente! Gracias, Obijuan ;)

closed 2021-04-04 by sasasuio