FPGAwars/icestudio #493: toolchain install stuck at 50% (install default apio..)

jojo535275, 2021-04-08:
hello everyone,

I got an colorligth i5 FPGA board and want to try to use it with icestudio.
So i just installed icestudio (0.5.1w210224-) on min Win10 . 
And when i launch the app, i can select a board (i tryed with various boards) 
Then i can launch the toolchain install. but this function is stuck at 50% (install default apio..)

Maybe it is only on my machine!? Any ideas to help me solve this issue

Thanks 

joaquim

Obijuan, 2021-04-09:
There seems to be a bug on Win10 that causes this problem (the reason why it happens it is still unknow as the users has not given us enough feedback). Please, try to install the toolchains manually following these instructions:

https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio

Any additional feedback will be very welcome in order to know what was the problem (missing dependencies, python issues...)

cavearr, 2021-04-09:
Hi!, review this howto :

https://github.com/FPGAwars/icestudio/issues/494

and attach to logging file.

jojo535275, 2021-04-09:
hello Obijan and cavear,

Thanks for your repplies.

some infos thay may help understanding  (or not :-) ) this issue:
1) i also tried to install it on a linnux virtual machine (lubuntu) and got the exact same  issue.
2) Then i installed (on win10) the previous version (icestudio-0.5.1n210107-win64)
and with thaht version i had no problem with the toolchain install.
Then, i installed the last version (0.5.1w210224) and there was no notifications to "install the toolchain" but i launched a toolchain update and it went fine.
So i think i can not reproduce the problem on win10. (or maybe by deletin manually)
3) But as i still can reproduce the problem on linux VM; here is the log file
[icestudio-log.txt](https://github.com/FPGAwars/icestudio/files/6284447/icestudio-log.txt)

Next, i will try to install the toolchain manually on linux VM and let you now the results.

muchas gracias

jojo535275, 2021-04-09:
I just tried the manual install of the toolchain on linux VM and it did the trick!

So everythink works fine here. i will now configure icestudio to work with my ecp5 fpga (colorlight i5 v7) following Fernando Mosquera [instructions.](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/H2LMoy7DE5s)

Thanks again guis for doing this great work.

closed 2021-04-14 by jojo535275