FPGAwars/icestudio #494: How to report debug information

cavearr, 2021-04-09:
When you find a bug that involves strange behavior and no error message appears, for example if icestudio hangs, please attach the following log file to the issue.

First, activate logging.  For this , go to **Edt -> Preferences**  and _click_ at **_Logging file_**

[image]

In the input box, write the path for the file that you want to use to logging information, for examle:

_/home/user/icestudio.log_ if you are on linux and _user_ is your username
or
_c:\icestudio.log_ if you are on windows (you could specify any directory)

[image]

Finally, return to **Edt -> Preferences**  and _click_ at **Logging enabled**

[image]

restart Icestudio, and do the action that produces the bug, close icestudio and send the logging file attached to the issue

Strauss77, 2022-03-30:
Hola soy un aficionado, me he comprado una Alhambra II me he bajado el icestudio 0.9.0 y me funciono sin problemas  me reconoció la placa y le cargué el primer ejemplo y sin problemas pero me di cuenta que en el menú no me aparecía "Básico" y he reinstalado, a partir de ahí me da un error de carga y la tarjeta ya no carga ningún esquema de icestudio. me podeis ayudar con esto? no se por donde seguir.  un saludo a todos
[image]

Obijuan, 2022-04-02:
Hola @Strauss77 

Hay 3 problemas. Vamos uno a uno

## Problema 1:

Has tenido un poco de mala suerte. La primera vez que instalaste Icestudio 0.9 todo te iba bien. El menú Basic se ha llevado a un panel nuevo accesible desde la opción Editar/Herramientas (o pulsando Ctrl-T)

[image]

Al pinchar en esa opción (o pulsando Ctrl-T en el teclado) te aparecerá el panel con las herramientas básicas. Se trata de un panel móvil que puedes colocar en cualquier parte (porque se usa mucho)

[image]

Por favor, confírmame que puedes ver el panel de Herramientas siguiendo estas instrucciones (para dar por concluido el problema 1)

## Problema 2: Mensaje de error al sintetizar el circuito

El 28 de Marzo, uno de los paquetes de los que depende el sintetizador de icestudio se actualizó a una version nueva, que provoca ese error

La mala suerte que has tenido es que desinstalaste Icestudio (que tenía el sintetizador correcto) y al volver a instalar Icestudio ya se te incluyó el nuevo sintetizador que causa el error.

Mientras sacamos una nueva versión estable de icestudio que solucione automáticamente el problema, te adjunto aquí la solución manual. Pincha en la opción Herramientas/Toolchain/Actualización (Última estable)

[image]
 
Se se intalará la toolchain con el error corregido. Si miras ahora en Herramientas/Toolchain te debe aparecer la nueva versión (compruébalo:)

[image]

Y ya podrás sintetizar tus circuitos (Para cargar en la Alhambra-II debes instalar el driver, pero eso ya lo hiciste la primera vez y por tanto sabes cómo se hace)

Por favor, confírmame que estas instrucciones solucionan el problema 2

## Problema 3:

No te aparecen las imágenes en el circuito hola mundo

Este es un bug conocido que aparece al instalar la toolchain la primera vez. Una vez instalada, cierra icestudio y vuelve a abrirlo. A partir de ese momento ya verás siempre todas las imágenes con normalidad (siempre)

(Al instalar la toolchain se borra el directorio de la caché. Y al arrancar Icestudio se crea, solucionándose el problema

Por favor, confírmame que estas instrucciones solucionan el problema

Para el resto de problemas nuevos que te aparezca, por favor, escríbelo mejor en el grupo de FPGAwars

Obijuan, 2022-04-02:
Hola @Alfredo2591 

Mira la respuesta que he puesto a @Strauss77 para solucionar los problemas. Antes de reportar nuevos problemas, por favor confírmame que se te han solucionado los mencionados

Alfredo2591, 2022-04-02:
!PERFECTO!
Los 3 puntos se han solucionado de maravilla.!
Gracias Juan por tu apoyo y dedicacion.

Obijuan, 2022-04-02:
Estupendo! Muchas gracias! 😃