FPGAwars/icestudio #498: Ecp5 bitstream file not updated after a pin change (as only change)

jojo535275, 2021-04-14:
Hello,

Fernando Mosquera and myself found an icestudio bug that affect the  ECP5 boards(and not the ice40 boards).
With our ColorLight board if we only change the pins in one project, the changes don't update the BIT file.
 We need to change anything in the project (add a new block or change something in the structure of the project), the only assignation or change of pins does not change the BIT file and the board is programmed with the first BIT file before the pin changes.

For exemple in my case:

[image]
[4 bit counter.zip](https://github.com/FPGAwars/icestudio/files/6308520/4.bit.counter.zip)

With the above simple 4 bits counter circuit, if:
-          I make an verify/build/upload
-          And I just make changes in the ouput numbers. (eg  P2_L18  becomes P2_L20)
-          I make second verify/build/upload 
Then my logic analyser does not see any wiring change on the ouputs signal. 
So it seems thaht the second upload was identical to the first one!

To make it work as expected I had to add a step in the sequence:
-          I make an verify/build/upload
-          I make changes in the output numbers. (eg  P2_L18  becomes P2_L20)
-          I add block (eg: constant block left unconnected on the workspace)
-          I make second verify/build/upload 

Then, my logic analyser see all the wiring changes on the output signals. 

Thanks to all the team for your work on icestudio and the collection libraries

Obijuan, 2021-08-09:
Thanks for all the information provided! 😀️

I've already fixed that in Apio. However It still cannot be easily tested in Icestudio. I will write the updates in this issue

Obijuan, 2022-05-02:
I think this issue has been fixed. @jojo535275: Can you confirm it using the latest wip? Thanks

jojo535275, 2022-05-03:
Hola juan, yes, i confirm, it is fixed !
Thanks

closed 2022-05-03 by jojo535275