FPGAwars/icestudio #499: Icesutudio doesn't swhow the large name of the ECP5's pins

benitoss, 2021-04-16:
Due to the large number of the pins in the new ECP5 boards, the pin's names has to be larger.
Actually Icestudio only shows 5 characters in the pin's field.
Please could you allow to show names with 6 characters (as example P2_C14) ???
Thanks

jojo535275, 2021-04-16:
i will add a precision on what Benitoss wrote.

the characteres are correctly shown on the screen ( even with 10 characters). But only the 5 first are saved. Therefore, when re-oppening a file (and maybe also when building it) a pin "P2_C14" will be mixed with a pin P2_C1.

What i saw is (for exemple) on the Alhambra II board
When you save the "SDA"  pin it reopens as "ADC_SDA" pin

You can open the joined file /change the pin to SDA/ then close and reopen it
[test.zip](https://github.com/FPGAwars/icestudio/files/6326817/test.zip)

and modifying the  "SDA" to "SDA_" in the pinout.json file solves the problem

jojo535275, 2021-04-16:
Little precision i could not saved the pinout.json for Alhambra II myself, because it is protected by the autors.
But for the unprotected pinout.json files like for the colorligth boards made by Benitoss, it fixes the problem.

jojo535275, 2021-04-16:
Little correction two :-)

In fact, for me, the problem is more that  the string "SDA" is in also in "ADC_SDA"

cavearr, 2025-01-24:
@lmcapacho I'm reviewing the list of bugs and new features to align with my current roadmap. I'm working on some spec definitions so that collisions don't occur in collaborations and we can organize ourselves better (IO dialog we talked about a few days ago and other things).

If you want to catch this bug/feature that would be great, it's a small thing but important because many times it's annoying.
It not only affects IO definitions, but also labels, names are cut off when they are large.
The goal should be to dynamically adapt the length size of each IO block or tag to the size of its name.

Now all IO blocks and labels are the same, after this task all labels must have a minimum size (the current one) and if the name is larger, the label or block must be larger.

If you don't want to collect, tell me without problems!

lmcapacho, 2025-01-25:
Hi @cavearr ,

I managed to catch this bug/feature. I need a bit more clarity about it. Could you let me know if the solution to this bug is as I show in this image?

[image]

cavearr, 2025-01-25:
Hello @lmcapacho! Thanks for taking on this feature!

The labels you show me are very well formatted, good job!

On the other hand, the problem appears in the *input* and *output* blocks, so there are two important points to rework:

1) First the selector block, in this case we could simply increase the width fixedly. I think the biggest names are on the cyntion board (you can select between the boards to see it). With a fixed wider value we can ensure all the names are readable:

[image]

2) Once the input or output has been selected, if it is larger than the current fixed width it cannot be read well, this width should be dynamic like in the labels you have already made.

[image]

Once again ,thanks for your work!

closed 2025-01-27 by cavearr