FPGAwars/icestudio #502: That particilar example (the S-R latch) does not work by default with the new toolchain (it is not a problem related to IceFun board). It worked ok with the old arachne-pnr tool, but when it was upgraded to nextpnr it stoped working. This new tool, by default, does not allow to have combinational loop. Of course, this can be overrided using the --force or --ignore-loops flags when invoking nextpnr:

ph1lj-6321, 2021-04-22:
That particilar example (the S-R latch) does not work by default with the new toolchain (it is not a problem related to IceFun board). It worked ok with the old arachne-pnr tool, but when it was upgraded to nextpnr it stoped working. This new tool, by default, does not allow to have combinational loop. Of course, this can be overrided using the --force or --ignore-loops flags when invoking nextpnr:

https://github.com/YosysHQ/nextpnr/issues/224

But this option has not yet been incluided in apio/icestudio

The S-R latch example should be removed or at least we should provide comments explaining that it does not work with the current toolchain (the --ignore-loops flag should be used)

_Originally posted by @Obijuan in https://github.com/FPGAwars/icestudio/issues/470#issuecomment-824541279_

ph1lj-6321, 2021-04-22:
How can I add this --ignore-loop flag ?

Obijuan, 2021-04-22:
It is not possible currently from icestudio. You have to do it from the command line by using directly the toolchain

ph1lj-6321, 2021-08-08:
Oh - can you please give me an example of how I can override this, I'm using 0.6.0 now with the IceStick Eval kit, module.

benitoss, 2021-12-10:
If you are in windows, you can add the "--force" option in line 177 in this file
"**C:\Users\_[your_user_in_windows]_\.icestudio\venv\Lib\site-packages\apio\resources\ecp5\SConstruct**"
For me the example S-R latch now is working

jojo535275, 2021-12-14:
Ola Benitos,

Thanks for that tip!
I also just managed to build the S-R Latch example and load it on my ecp5 board,  with success !
Do you see a reason for not pull-requesting this change  ?

Obijuan, 2021-12-14:
Hi @jojo535275 !

The pull-req was already issued by Fernando and merged in apio:

https://github.com/FPGAwars/apio/pull/257

It is currently in the apio development branch. You can try it by installing the development toolchain in icestudio (but as it is a develop version, other stuff can stop working. Use with caution. You can go back to stable at any moment)

Apio is in the process of uodating the toolchains. Many things are broken and should be fixed before it can be integrated as stable into icestudio)

jojo535275, 2021-12-14:
Ola Juan, 
Thanks for the info,
I will try to use these develop version of apio. and report the bugs, if i found some!

tlerner-pizza, 2022-09-29:
I tried building an SR latch circuit and it gave me an error as well. it looks like the --force is already in "C:\Users_[your_user_in_windows]_.icestudio\venv\Lib\site-packages\apio\resources\ecp5\SConstruct" but it still gives me the error "ERROR: timing analyss failed due to presence of combinatorial loops, incomplete specification of timing ports, etc."

mgesteiro, 2023-02-04:
> I tried building an SR latch circuit and it gave me an error as well. it looks like the --force is already in [...]

What board are you using @tlerner-pizza ? maybe you have an **ICE40** family model instead of the **ECP5**, like it was the case for @ph1lj-6321 in the previous comment (or for me).

If that's the case, you should edit the "C:\Users_[your_user_in_windows]_.icestudio\venv\Lib\site-packages\apio\resources\ice40\SConstruct" file instead, and add the extra `--ignore-loops` parameter like this:

`    action='nextpnr-ice40 --ignore-loops --{0}{1} --package {2} --json $SOURCE --asc $TARGET --pcf {3} {4}'.format(`

tlerner-pizza, 2023-02-07:
That was indeed the issue. I assumed it was in both since it was already in ECP5, but I should have checked.

mgesteiro, 2023-02-07:
Welcome! Maybe @ph1lj-6321 (or the repository admins) can mark this issue as solved and close it 👌

ph1lj-6321, 2023-02-07:
Hi

I dont seem to be able log in any more - perhaps Admin can close it

Philip Jarvis
Professor,
School of Science and Engineering Technology
Electronics Department
Durham College
(905) 721-2000 x2467

[EXTERNAL EMAIL - USE CAUTION]

This message is intended only for the named recipients. This message may contain information that is confidential or exempt from disclosure under applicable law. Any dissemination or copying of this message by anyone other than a named recipient is strictly prohibited. If you are not a named recipient or an employee or agent responsible for delivering this message to a named recipient, please notify us immediately, and permanently destroy this message and any copies you may have. Warning: Email may not be secure unless properly encrypted.

cavearr, 2023-02-07:
Thanks team! close the issue, i'm working in do this kind of things easier (settings custom options for apio) in the next weeks i'm releasing new wips with it.

Thanks for your support!!!

closed 2023-02-07 by cavearr

mgesteiro, 2023-02-07:
ah, ok. Let me know please, I was actually working in the same...

cavearr, 2023-02-07:
I'm reworking part of the core in the way of decoupling Apio, to allow the use of other backends in the future (like fusesoc or commercial tools). This new approach will allow passing custom arguments in each layout.

I have little left but at least until the first week of March I don't think I can publish the first wips.

I don't know what approach you're taking if you want to share your thoughts ahead!

mgesteiro, 2023-02-08:
My plan was to include the possibility to add extra parameters per Icestudio project (i.e. specific to a design instead of a "global" option) that would be propagated to APIO and -finally- to the low level tools. This way, projects/designs like the SR-Latch, that require special options, would work. Makes no sense now if you are already decoupling those parts (you'll sure do a better job as you know better the whole system!).
Is there any document or place where we can see what are you working on, or what the future plans are? I have other ideas and it would be nice to coordinate efforts instead of colliding 👌

cavearr, 2023-02-08:
Hello @mgesteiro! I am very happy to see your motivation with Icestudio, first of all, thank you!!!

I'm preparing many new things for the coming months, very much in line with your idea. I'm actually writing a dev wiki, but I need to close some major changes that redefine the core and plugins before I publish it, because otherwise it's going to be even more chaotic.

 When I finish it I will publish a very detailed roadmap for the next months, in fact I will ask for help in the community because I want to do a lot of strength in collecting information, tutorials, etc.

If you want to help me, there are two tasks, one is related to the reason for this thread. The idea would be a new panel for configuring, modifying and creating new boards (actually the process is very cumbersome at the moment), in this new panel/window there should be an area to define custom build commands eg. If you want to play around with this, let me know and we could create an issue with this and discuss the whole process, ideas, architecture, etc. over there.

Another imminent task that is needed is to migrate the current graphics library used in Icestudio to its new version which is not 100% compatible and the code needs to be updated to substantially improve performance in order to continue working with the rest of the updates.

Another important task is related to the component library (improving functionalities such as search, information panel, better loading...)

I don't know if you want to help with any of these, but if you like it, please let me know!

mgesteiro, 2023-02-08:
> Hello @mgesteiro! I am very happy to see your motivation with Icestudio, first of all, thank you!!!

It's a nice tool, I'm the grateful one 👌

> I'm preparing many new things for the coming months, very much in line with your idea. I'm actually writing a dev wiki, but I need to close some major changes that redefine the core and plugins before I publish it, because otherwise it's going to be even more chaotic.

major changes are always difficult: after many hours working in a project, sometimes is necessary to go back to the drawing board, this time with the knowledge of the previous experience, right? I see many things that could use some help to dramatically improve UX, but I recognize also that -*from outside*- (even for an experienced developer) it always seems easier... So, all my support for you!
 
> When I finish it I will publish a very detailed roadmap for the next months, in fact I will ask for help in the community because I want to do a lot of strength in collecting information, tutorials, etc.

nice
 
> If you want to help me, there are two tasks, one is related to the reason for this thread. The idea would be a new panel for configuring, modifying and creating new boards (actually the process is [...]

I see. Is not the most interesting part for me, but happy to help.

> Another imminent task that is needed is to migrate the current graphics library used in Icestudio to its new version which is not 100% compatible and the code needs to be updated to substantially improve performance in order to continue working with the rest of the updates.

this part is more interesting for me. I could take the opportunity to dive-in into this part: my knowledge of the system is quite high-level/superficial right now.

**Are you talking about JointJS here** or something else?
 
> Another important task is related to the component library (improving functionalities such as search, information panel, better loading...)

projects/design handling could use some improvement also (e.g. there is no "close project" and everything is a new-window).
 
> I don't know if you want to help with any of these, but if you like it, please let me know!

I'm willing to get involved (no compromise, though). I'll continue to explore subsystems in the system, getting familiar with it. Feel free to point me to somewhere specific...

cavearr, 2023-02-08:
With no obligation!! If working on jointjs motivates you, go ahead with this task. At the moment my concern is migrating to the new jointjs, and this task is taking me every time and it is very important to carry it out.

if you want to take it, when you finish it, tell me what kind of tasks you are interested in and I will think about things around.

If you want to create a problem for this task, such as "Rendering engine update" and there we discuss the progress, doubts, ideas, etc.

All your ideas are welcome and you can develop as many as you want. In fact, if you have any new functionality in mind, the ideal is to do it as a plugin since we will soon have a kind of "store" where we can upload plugins, install them easily, etc. I'm telling you this because it still motivates you more to do some new functionality that you have in mind, whatever you prefer!

Thank you so much!