FPGAwars/icestudio #504: Memory block error: Can not open file for readmemb

jdeitmerg, 2021-06-14:
Hi,

the memory block doesn't seem to work.
* its contents are written to a .list file in the temporary build folder
* $readmemb/$readmemh apparently sees an empty argument instead of the path to the .list file

Attached is the following example, along with my log file:
[image]

Console output:

```
[Mon Jun 14 12:26:40 2021] Processing icezum
--------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q main.v
main.v:35: ERROR: Can not open file `` for \$readmemb.
scons: *** [hardware.json] Error 1
========================= [ ERROR ] Took 0.97 seconds =========================
```

[memory_block_issue.zip](https://github.com/FPGAwars/icestudio/files/6647670/memory_block_issue.zip)

jdeitmerg, 2021-06-14:
As the log file doesn't contain the version of apio: I just updated the toolchain, the problem persists.

Obijuan, 2021-06-14:
You should use the DATA parameter like this:

```verilog
if (DATA) $readmemb(DATA, test);
```

During the synthesis the DATA parameter is "". Its value it is no assigned until  the elaboration phase, were the memory is initialized. To avoid any error with the sinthesizer you have on add that "if" sentence

You can have a look at the [icemem collection](https://github.com/FPGAwars/iceMem). In the SMemory menu you will find some working memories

An attaching your example fixed

[memory_test-fixed.zip](https://github.com/FPGAwars/icestudio/files/6649096/memory_test-fixed.zip)

jdeitmerg, 2021-06-14:
Oh, I see. Thank you!

closed 2021-06-14 by jdeitmerg