FPGAwars/icestudio #505: Can't install due to adam-lynch/node-temp missing package.jso

Phil67ago, 2021-06-15:
Hi!

[~]$ git clone https://github.com/FPGAwars/icestudio.git
[~]$ cd icestudio
[~]$ npm install
npm ERR! code ENOLOCALFrom:

npm ERR! Could not install from "node_modules/nw-builder/temp@github:adam-lynch/node-temp#remove_tmpdir_dep" as it does not contain a package.json file.

I'm not so familiar with npm but https://github.com/adam-lynch has no 'node-temp'???

npm version: npm-6.14.13-1.14.17.0.1.fc34.x86_64

Obijuan, 2021-06-15:
Make sure you have installed the correct node version (it should be 10.17.0)

You can find more information on this link: https://github.com/FPGAwars/icestudio/wiki/Icestudio-development-version

Phil67ago, 2021-06-16:
I solved it by deleting 'package-lock.json'
[user ~]$ cd icestudio
[user ~]$ rm package-lock.json
[user ~]$ npm install
launch:
[user ~]$ npm start

node version: nodejs-14.17.0-1.fc34.x86_64

closed 2021-06-16 by Phil67ago