FPGAwars/icestudio #506: tool install error in Win10

jimprince2017, 2021-06-21:
After I set APIO_HOME_DIR
I get: set APIO_HOME_DIR="D:\SPB_Data\.icestudio\apio"& "D:\SPB_Data\.icestudio\venv\Scripts\apio.exe" install system

I do not know how to interpret & "D:\SPB_Data\.icestudio\venv\Scripts\apio.exe" install system

The error window is cut-off and I can not read it.

Obijuan, 2021-06-23:
Could you please try the lastest build image?: 0.5.1w202106230206
You can download it from here: [downloads.icestudio.io](https://downloads.icestudio.io/)  
We have fixed some problems related to windows installation

Obijuan, 2021-07-05:
We have released the 0.6.0-rc2 icestudio version. You can download it from the [Release section](https://github.com/FPGAwars/icestudio/releases). It would be great if you could install it and test it. After installing the icestudio, please remove the previous toolchain (menu option tools/toolchain/remove) and intall it from scratch. 

We will release the final stable version (0.6.0) by July 10th. If there is no feedback by then, we will considered the problem is fixed and we will close this issue (as other win10 users have not reported any problem)

We are looking fordward to hearing from you
Thanks

Obijuan, 2021-07-13:
The [icestudio 0.6.0 version](https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0) has been released. Many users have reported that it works ok regarding the toolchain installation. Please try it and you the problem remains, feel free to reopen this issue again

closed 2021-07-13 by Obijuan