FPGAwars/icestudio #507: no usable home directory

R-Hidalgo, 2021-06-22:
Estimados,

Desde ya muchas gracias por su colaboración,

Al instalar icestudio-0.5.1w202106211106-win64.exe  en Win10 Pro x64 no se instala la toolchain. 
Al usar la instalación externa de la toolchain tampoco se soluciona el problema ( https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio#windows-10)

Adjunto características del sistema, mensaje de error y archivo de log.

[image]

[image]

[image]

[image]

[log_icestudio.txt](https://github.com/FPGAwars/icestudio/files/6690420/log_icestudio.txt)

R-Hidalgo, 2021-06-22:
Estimados,
Agrego lo siguiente respecto del problema:

Icestudio cambió su carpeta desde .icestudio a icestudio_home . ( con espacio al final).
Luego, no es posible aplicar la instalación externa de la toolchain y además Icestudio duplica su carpeta.

[image]

[image]

cavearr, 2021-06-22:
Hi!, uninstall this wip and try the new one icestudio-0.5.1w202106221206-win64.exe, the wip that you install was an error.

To delete the folder with the white space at the end, you could execute in the cmd:

rd /s "\\?\c:\Usuarios\gusta\icestudio_home "

Obijuan, 2021-06-23:
@R-Hidalgo muchas gracias por el feedback tan detallado. Nos aporta mucha información y gracias a ello podemos solucionar los problemas  
¿Puedes probar la última imagen generada? 0.5.1w202106230206  
Debería instalar la toolchain correctamente sin ese error. Para asegurarnos que todo va bien, sería genial si una vez instalada puedes eliminarla (Herramientas/toolchain/eliminar) y volver a instalarla. Me quiero asegurar que efectivametne está todo ok, antes de pasar a solucionar el resto de bugs

Muchas gracias por tu ayuda 😀️

R-Hidalgo, 2021-06-23:
Gracias Obijuan, 

Desinstalé la versión anterior y eliminé los directorios erróneos.

Descargué e instalé la última versión icestudio-0.5.1w202106230206-win64.exe. Pero al instalar la toolchain se produjo otro error:

[image]

En icestudio.log se encuentran estas líneas con el error:
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\apio\managers\installer.py", line 124, in install
    makedirs(self.packages_dir)
  File "C:\Users\gusta\AppData\Local\Programs\Python\Python38\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 3] El sistema no puede encontrar la ruta especificada: 'C:\\Users\\gusta\\.icestudio\\apio \\packages'

Luego, seguí los pasos para hacer instalación externa de la toolchain. Pero al iniciar Icestudio nuevamente, no reconoce la toolchain, indica que no está instalada.

R-Hidalgo, 2021-06-23:
Acá está el bloque completo del log con el error, con el doble back slash al final:

>>(ERROR): Traceback (most recent call last):
  File "C:\Users\gusta\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\gusta\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\gusta\.icestudio\venv\Scripts\apio.exe\__main__.py", line 9, in <module>
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\click\core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\apio\commands\install.py", line 50, in cli
    Installer(package, platform, force).install()
  File "c:\users\gusta\.icestudio\venv\lib\site-packages\apio\managers\installer.py", line 124, in install
    makedirs(self.packages_dir)
  File "C:\Users\gusta\AppData\Local\Programs\Python\Python38\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 3] El sistema no puede encontrar la ruta especificada: 'C:\\Users\\gusta\\.icestudio\\apio \\packages'

R-Hidalgo, 2021-06-23:
No, no, tampoco lo muestra. Es el editor de Github que quita el doble back slash de la última línea.

R-Hidalgo, 2021-06-23:
Adjunto el archivo Icestudio.log, tiene 2 días de registros. 

[icestudio.log](https://github.com/FPGAwars/icestudio/files/6698833/icestudio.log)

Obijuan, 2021-06-23:
El error es debido a que se obtiene un path con un espacio al final. Eso ocurría en la imagen anterior 0.5.1w202106230106. Unos minutos después he generado una imagen nueva que ya no incluye el espacio. Fíjate que son muy parecidas, solo se diferencian en un número: 0.5.1w202106230206

Esta es la imagen que he probado en un win10 y va bien (en la anterior me salia el mismo error que a ti).  Necesito que borres todos los icestudios que hayas descargado y que te asegures de instalar    0.5.1w202106230206. Antes de arrancarlo borra el icestudio.log para que se genere uno nuevo. Comprueba también en Ayuda/version que efectivamente se ha cargado el icestudio version  0.5.1w202106230206

Necesito que hagas esto para asegurarnos al 100%. Yo ahora estoy de camino a la universidad y lo probaré en otra máquina windows

En las próximas imágenes vamos a meter la version de icestudio en el log (pero de momento se nos ha olvidado)

Muchísimas gracias por tu ayuda. Este tipo de feedback y pruebas son vitales! (Y pocos usuarios nos dan tan buenos detalles. Así es muchísimo más fácil encontrar los bugs)

R-Hidalgo, 2021-06-23:
Gracias Cavearr, gracias Obijuan, ahora sí cargó la toolchain con icestudio-0.5.1w202106230206-win64.exe

Después eliminé la toolchain desde Icestudio,  para volver a instalar desde el menú. Hizo nuevamente la instalación sin problemas.
[icestudio.log](https://github.com/FPGAwars/icestudio/files/6699185/icestudio.log)

[image]

closed 2021-06-23 by R-Hidalgo

Obijuan, 2021-06-23:
Bien!!!!!!!! Muchísimas gracias!!!!! Qué alegría da cuando se soluciona un bug  
Ahora a por el siguiente!!! :-)

Obijuan, 2021-06-23:
Lo acabo de probar en otro ordenador Windows de la Universidad y confirmo que funciona correctamente. Ya se ha probado en tres ordenadores windows 64-bits diferentes y va ok