FPGAwars/icestudio #512: Toolchain no install_win64

fran-minas, 2021-07-01:
Hi Obijuan,

I'm try to install the toolchain in new icestudio version (icestudio-0.5.1w202106261206-win64), but the installation bar freezes at 30% (the computer has Python 3.9 version).
I did not have problems with Icestudio earlier versions. This is the first time I check this "problem".
I attach a screenshot about it.
Best regards!.
[image]

Obijuan, 2021-07-01:
Hi fran-minas!
Thanks for the feedback. I need you to test something, in order to have more information about this bug
Could you please remove the toolchain (click on tools/Toolchain/Remove) and then Install it again (Tools/Toolchain/Install (Stable))

Then Tell me what happened. If the error persists i will ask you in this issue to do more tests

fran-minas, 2021-07-01:
Hi Obijuan,

the problem remains the same (freeze at 30% in the install bar)...I attach the *.png about it.
I remember that this install take a few seconds other times...now it can be in this percentage more than hour (I tried to let it install but I assumed that so long is not normal and closed the software).
Thanks for all!!..
[image]

fran-minas, 2021-07-01:
Here it is. The System info:
[image]

Obijuan, 2021-07-01:
ok, Thanks! I can see were the potential error could be. Your user path contains an accent (tilde): "C:\Users\FranciscoJosé". It should work ok anyway, so let's try to find were the bug is

I need you to follow these steps:

1) Remove the file C:\Users\FranciscoJosé\icestudio.log (it is a log file create by icestudio)
2) Run icestudio
3) Install the toolchain (and wait 2 or 3 minutes after it reaches the 30%) 
4) There should be a new log file: C:\Users\FranciscoJosé\icestudio.log.  Attach that file into these issue

I will read this text file carefully to detect the error and confirm that it is due to the `é` character. Then I will try to reproduce it and fix it

Thanks a lot for your feedback. It is really helpfull for improving icestudio! :-)

fran-minas, 2021-07-01:
Ok!.
I already followed the steps (same freeze at 30%...10 minutes install wait at that precentage).
I attach the new created "icestudio.log" at 01/07/2021_20:58.

Thanks for you attention!!
Best regards, Fran.

[icestudio.log](https://github.com/FPGAwars/icestudio/files/6750283/icestudio.log)

Obijuan, 2021-07-01:
Thanks! IT is strange. There is no error message shown in the log, but the installation is stalled. Let's try to execute the command directly on the command line, to see what happens

Could you please open the windows cmd (command line) and execute this command?

`"C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3" install -U apio[blackiceprog,tinyfpgab,tinyprog,icefunprog]==0.6.7`
(including the quotes. Just copy and paste it and take a screenshot. It is a long command in one line, although in this issue is displayed in two lines)
(I am really curious about this error)

fran-minas, 2021-07-01:
....nothing has happened. It may be an error to literally copy the command over this header. When I start "Windows_símbolo del sistema"...the command line start as: **C:\Users\FranciscoJosé>** (it is not possible to delete...)
The screenshot about it is next:
[image]

Obijuan, 2021-07-02:
ok, Thanks! You have executed the command perfectly!

What is strange is that the command does not produce any output... It is the first time I see something like this. Let's execute the same command but without arguments, to try to understand what is happening (I wish i have your computer here for testing! hahah)

Could you please execute in the windows cmd the following commands and post the screenshots?

* First command: `"C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3"`

* Second command: `"C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3 --version"`

fran-minas, 2021-07-02:
Good morning Juan,
at this moment, the concept to run and check icestudio overtakes my knowledge... : (
This problem and my computer is at your complete disposal... ; )

- First command: "C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3"
[image]

- Second command: "C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3 --version"
[image]

Obijuan, 2021-07-02:
ok, thanks! These command are not about icestudio, but python. There is something strange in your computer related to python and the virtualenv. If it has happened to you, it could happen to other userstoo, so I am gathering information to discover what the problem is

These are the new commands I want you to try:

* First command: `"C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3" --version`    
* Second command: `"C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip" --version`  
* Third command: `dir "C:\Users\FranciscoJosé\.icestudio\venv\Scripts"`

fran-minas, 2021-07-02:
New screenshots:

- First command: "C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip3" --version
[image]

- Second command: "C:\Users\FranciscoJosé\.icestudio\venv\Scripts\pip" --version
[image]
 
- dir "C:\Users\FranciscoJosé\.icestudio\venv\Scripts"
[image]

fran-minas, 2021-07-02:
The toolchain is installed Juan!!!.....

After your last command proof recommendation, I had restarted the PC because it slowed down....i executed the steps....and send the screenshots......but, but, but, I tried executed Icestudio and install the toolchain, and at this moment....THE ANTIVIRUS RUN!!!! and detected the "icestudio.exe" as a possible virus. I added the *.exe in a exclusion list, and the toolchain install was completed.
The problem was the antivirus-firewall, that blocked the icestudio execution, and I did not get any alerts about it.

I feel st**id, and I apologize for the time that I have wasted you...,but I had never had this problem between the icestudio executable and the antivirus (and the toolbar alert has not warned me of it).

[image]

Obijuan, 2021-07-02:
Goood news!!!! You do not have to feel bad about it. On the contrary, you are providing a very useful information!! What has happened to you, could happen again to anyone. For that reason is important to retrieve all the information.

Could you give more details on how you fixed the issue? I will write it down in the Icestudio FAQs

I need to know which antivirus were you using (name, version...)  and more details about how to remove icestudio.exe from the blacklist

Thanks a lot 😀️

fran-minas, 2021-07-02:
The antivirus is AVAST (I attached screenshot with last info software).
[image]

Steps to solve the problem:
1) Run "icestudio.exe" (nothing weird detected).
2) Select install toolchain: at 10% the AVAST was activated automatically (automatic new advise in my toolbar) about _"POSIBLE SOFTWARE MALICIOSO SE ESTÁ EJECUTANDO"._
3) AVAST gave me 2 options: remove the app (as a security measure) or trust it and add the "icestudio.exe" to the exclusion list.
4) Select "include in the exclusion list", icestudio run and the toolchain install continued and finished satisfactorily.

The strange thing is that it didn't happen the first time I ran icestudio.exe, which was not letting me see the problem first. 
I got a hunch when I restarted my computer to try again and that's when it all happened.

Juan, 
Thank you very much for your time and attention. I have been following you few years ago about 3D design with Freecad and I was introduced myself to the world of FPGA's programming through your tutorials. It has also been very useful for me in my classes with my students.

My sincere congratulations for all job done. It has been a pleasure to establish this contact with you.
May free software be with you!! ; )

Best regards,
Ph.D. Eng. Francisco Carnero (Asturias/Spain)

Obijuan, 2021-07-02:
Hi Francisco!

Thank you very much for the information and your kind words 😀

I will publish it as part of the FAQs after the weekend (with credits to you, of course). Please keep this issue open until I accomplish that

I will add the link to the FAQs on this issue when i am done

Es un placer colaborar con gente como tú. Muchísimas gracias por la información. Así conseguiremos hacer una herramienta mejor entre todos 😀

fran-minas, 2021-07-02:
Perfect!.
I will keep this issue open in case you need something else, do not worry about it.
I am glad I made a small improvement step to this system.

Un placer haber conversado contigo y colaborado con la comunidad!! 😀.
Saludos y abrazos.
Ph.D. Eng. Francisco Carnero (Asturias/Spain)

Obijuan, 2021-07-05:
The problem and its solutions is alredy on the [FAQs](https://github.com/FPGAwars/icestudio/wiki/FAQs-and-troubleshooting)  
I will close this issue  
Thanks you very much for your feedback

closed 2021-07-05 by Obijuan