FPGAwars/icestudio #513: No Toolchain-Installation behind Firewall

hpsteidle, 2021-07-07:
We intend to use icestudion in a school-environment.  When we try to install the Toolchain from icestudio (nightly-build version), we get an error-message "no internet connection".

After analyzing the traffic on our Firewall, we discovered filtert dns-request packages. Further analytics showed , that the dns-requests tried to connect different dns-root-servers and not the local dns-Server..

After that, we tried the stable version of icestudio - and that worked!  Was there a change in the dns-behavion of the new version?

Obijuan, 2021-07-13:
The toolchains used by icestudio are installed using pip. One of the installed packages is [apio](https://github.com/FPGAwars/apio/wiki) which in turn connect to github for downloaling the binaries for your platform. There is nothing implemented in icestudio regarding the dns as far as we know. It will connect to pypi for installing the apio package and github for the binaries

The toolchains can also be installed manually, from outside icestudio. The instructions are on this link: [Installing the Toolchains from outside icestudio](https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio)  

I will close this issue. If you find any problem, please do not hesitate in reopening this issue

hpsteidle, 2021-07-17:
I was able to sucessfully install the Toolchain from outside icestudio. To get it working behind our proxy i had to set a environment variavle HTTPS_PROXY=http://**server-ip**:**port**.

It woud be nice, if this would also work within icestudio - i assume icestudio makes the same steps for installing the toolchain? When i try to install the toolchain in icestudio after setting the environment variable, there still comes the error "no internet connection". This error-hind appears also in the log-file of icestudio.
I took al closer look alt the sourcefiles of icestudio and realized in ../app/scripts/services/tools.js - line 1339 that the internet- connection is checked before installing the Toolchain by a function called isOnline(..). This function seems to belong to a nodejs-module "is-online". So far i understand, is this funktion not working behind a proxy out of the box..

It would be great, if there would be a setup-Entry in icestudio for using a proxy (which generates the environ variable for the proxy and skips the internet connection-test)

hpsteidle, 2021-07-17:
Hello Juan,
  unfortunately i dont know how to reopen the thread :-(
  So i only made a further comment..

Hans-Peter

>>> Juan Gonzalez-Gomez ***@***.***> 13.07.2021 08:35 >>>
The toolchains used by icestudio are installed using pip. One of the installed packages is [apio](https://github.com/FPGAwars/apio/wiki) which in turn connect to github for downloaling the binaries for your platform. There is nothing implemented in icestudio regarding the dns as far as we know. It will connect to pypi for installing the apio package and github for the binaries

The toolchains can also be installed manually, from outside icestudio. The instructions are on this link: [Installing the Toolchains from outside icestudio](https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio)  

I will close this issue. If you find any problem, please do not hesitate in reopening this issue

LBertrandDC, 2021-08-14:
Using icestudio 0.6.0 on Windows 10 Enterprise (installed by my college IT services), I was also having trouble with "No Internet connection". Following the hint from hpsteidle, I commented out the if statement after the call to nodeOnline and simply called callback(). It worked just fine. 

Line 327 of
C:\Program Files\icestudio\scripts\services\utils.js 

However on another machine running Windows 10 Pro the 0.6.0 installation went smoothly.

Is there anything you would like me to try?

cavearr, 2021-08-14:
Hii @LBertrandDC ! could you try with new release 0.7.0 or last wip ? and tell us if the same error appears?

i'm reviewing this online check , but if you could try the newer release or the last wip should be very usefull.

Thanks!

LBertrandDC, 2021-08-14:
Hi @cavearr I just tried it with 0.7.1w202108130708 and I still get the same error, "internet connection required". 

Also, I get an error at startup (is this because apio is not yet installed?):

set APIO_HOME_DIR="C:\Users\louis\.icestudio\apio"& "C:\Users\100159386\.icestudio\venv\Scripts\apio.exe" --version
The system cannot find the path specified.

LBertrandDC, 2022-05-02:
Hello again, I just tried with 0.9.2w202205020505 Windows 10 Enterprise 20H2 64-bit and I got the "Internet connection required" message.
Workaround in the usual ugly manner by commenting out the if statement in the function service.isOnline() (starts at line 215) that in turn calls nodeOnline(), and always returning success.
```
    this.isOnline = function (callback, error) {
      nodeOnline({
        timeout: 5000
      }, function (err, online) {
            callback();   
        // if (online) {
          // callback();
        // } else {
          // error();
          // callback(true);
        // }
      });
    };
```
Is there something else that you would like me to try?

cavearr, 2022-05-02:
Hi @LBertrandDC ! thanks for your feedback and efforts, i'll check  this nodejs test.

I'll try to reproduce it.

cavearr, 2022-05-02:
A question...
Are you behind a proxy or connect throught vpn?

What kind is your internet connection?

cavearr, 2022-05-02:
Could you uninstall icestudio and reinstall the latest wip 0.9.2w202205020905 from https://downloads.icestudio.io

and tell us if it works?

Thanks!

LBertrandDC, 2022-05-02:
Hi @cavearr, I just tried it on the same setup as earlier today and it installed the toolchain just fine. I tested with a little blinkie for the Lattice iCEstick.

Obijuan, 2022-05-03:
Great! I think this issue can be closed. Please, feel free to reopen it again if the error remains, or open a new issue in case new errors are found.
Thanks for the feedback

closed 2022-05-03 by Obijuan