FPGAwars/icestudio #517: macOS IceStudio 0.6.0 reported damaged

Gasman2014, 2021-07-13:
I have been successfully running IceStudio 0.6.0-rc2 for a couple of days. I noticed that you just tagged 0.6.0 so downloaded it (I want to try with the IceSugar Nano). 

Unfortunately, my Mac reports the file as damaged. (I have tried both the .zip and the .dmg).

[macOS 10.15.7 Catalina with Gatekeeper disabled]

Obijuan, 2021-07-13:
Thanks for the feedback! We will check it and confirm the bug.   
Version 0.6.0 is quite similar to 0.6.0-rc2 (just a small change), so you can continue working with 0.6.0-rc2 by now

Gasman2014, 2021-07-13:
Thanks. Actually I didn’t realise that the IceSugar was available in RC2 and thought I needed 0.6.0 - the board menu doesn’t have a scrollbar to indicate the there are additional boards available. I’m still waiting for my board to arrive.

Obijuan, 2021-07-14:
I cannot confirm the bug. It works ok for me. I have just downloaded the dmg file, installed it and it worked ok...

[image]

Could you please download it and try it again? (Just the dmg file is fine)

I will ask other mac users to try it

Gasman2014, 2021-07-14:
Restarted my Mac, downloaded the tagged version from here [https://github.com/FPGAwars/icestudio/releases/download/v0.6.0/icestudio-0.6.0-osx64.dmg](url) with Safari.  Dragged to /Applications directory.
Control click to open and I still get "“icestudio” is damaged and can’t be opened"

```
>md5sum icestudio-0.6.0-osx64.dmg
8ce9d07c514640b034027c0c159ddb94  icestudio-0.6.0-osx64.dmg

>sha1sum icestudio-0.6.0-osx64.dmg
a30801cf91dbb2bde89c99779e8aaa6f9f30b121  icestudio-0.6.0-osx64.dmg
```

I also downloaded the zip version with Firefox and the same thing happens. I've tried extracting with tar too - no difference.

Happy to work with rc2 for the mo.

Obijuan, 2021-07-14:
Thanks for the test. I've downloaded the .dmg file on my linux system and the checksum is the same as you

```
$ md5sum icestudio-0.6.0-osx64.dmg 
8ce9d07c514640b034027c0c159ddb94  icestudio-0.6.0-osx64.dmg

$ sha1sum icestudio-0.6.0-osx64.dmg 
a30801cf91dbb2bde89c99779e8aaa6f9f30b121
```
So I have no clue on what's going on. Both, the 0.6 and the rc2 have been automatically generated by the same github action script, and the changes between them are minimum

Now it is not a problem. The rc2 is fine for you, but it can also happens in the next stable version, so it is important for me to discover what is the problem

I will wait for the report of other mac user

Thank you very much for your tests

Obijuan, 2021-07-15:
Hi!, Other mac users has tested the file and it worked for them (It was tested on not Big sur macs). Why the rc2 is working ok for you but the 0.6 doesn't is still unknown. It can be due to the apple security mode (for preventing the execution of programs not signed by apple)... or it can be due to the execution of icestudio (generated for intel processor) in the M1 processor (in the case of Big Sur).
So, Let's close this issue by now. More people will test it and we will know the answer. If this error ocurrs in future stable version please feel free to reopen this issue or create a new one

Thanks you very much for your feedback

closed 2021-07-15 by Obijuan