FPGAwars/icestudio #519: Entre etiquetas no se crea el cable.

Democrito, 2021-07-24:
Hola Carlos,

Esta Issue que te voy a comentar es una tontería y no hace falta responder a este mensaje.

No se crea cable entre dos etiquetas, eso es todo. Es muy raro hacer esto porque en apariencia no tiene sentido hacerlo, pero mientras hacía un diseño, por simplicidad y de forma momentánea enlacé dos etiquetas y al cabo de un tiempo me di cuenta de lo que sucedía. Cree un ejemplo para asegurarme y eso ya lo confirmó. Adjunto imagen de ejemplo.

Un abrazo!

[image]

Obijuan, 2021-08-31:
La semántica actual es que las etiquetas sólo se pueden conectar a bloques, y no entre sí  
El poder conectar dos etiquetas entre sí, simulando un cable, es una característica que tiene muchas utilidades. Yo también lo he encontrado necesario

Para solucionarlo temporalmente he añadido el bloque "wire" a la [colección icewires](https://github.com/FPGAwars/iceWires) cuya función es justamente esa: Un bloque que sólo implementa un cable para poder conectar etiquetas

Añadiendo el bloque wire entre las etiquetas xx - yy del ejemplo que muestras, se consigue que funcione como se espera:

[image]

También funciona usando buses:

[image]

Estos bloques se encuentra en el **menú Wires** de la colección icesWires

Democrito, 2021-08-31:
Muy buena idea (la primera vez que lo vi no me di cuenta de ello), además los "wires" permiten mejorar el diseño ya que después de "torcer" muchas veces una conexión comienza a hacer cosas extrañas (en el sentido gráfico). Sea como fuere las etiquetas ha sido un paso crucial para poder diseñar circuitos complejos y me siento enormemente agradecido (gracias Carlos!).

cavearr, 2022-04-26:
Esta issue creo que está resuelta ¿la damos por cerrada?

Obijuan, 2022-04-27:
Todavía no se puede cerrar. Es una feature que hay que implementar. Lo que propuse aquí es una solución temporal

TimRudy, 2024-04-18:
I don't understand this (English translation). It seems that `xx` can be connected to `yy` with a wire (2 DIFFERENT labels). That is accepted. It is shown visually. But it does not actually connect them?

What's the desired feature for the user?
- Allow that visual connection (no change to current behaviour)
- Make the wire work as a wire
- Maybe create an alert when they are wiring: "Are you sure you want to connect 2 DIFFERENT labels"?

cavearr, 2024-04-18:
What democritus is referring to is using links between tags to simplify graphical connections as if they were a "hub".

That cannot be done right now, it would be a new feature to implement.