FPGAwars/icestudio #520: Aarch64 dependancies missing

janrinze, 2021-07-27:
Thanks to the work of many people finally a nw.js binary for linux-aarch64 is available at [https://github.com/LeonardLaszlo/nw.js-armv7-binaries/releases](https://github.com/LeonardLaszlo/nw.js-armv7-binaries/releases) .
Since this is the first time to have native linux-aarch64 version running icestudio it is unsurprising that we run in to some issues.

Installed on Debian Sid aarch64 using the following strategy:
- download and unpack the nw.js binary release in 
- clone the icestudio repository in directory "icestudioSRC"
- remove the nw dependencies in package-lock.json and package.json  (we have a binary now)
`cd icestudioSRC
npm install
cd app
~/<nwjsbinaryunpackdirectory/nw --ignore-gpu-blacklist --disable-accelerated-2d-canvas --num-raster-threads=2 ./`

it takes a little time to startup.

when installing the toolchain errors will appear.
I manually installed several packages:
`. ~/.icestudio/venv/bin/activate
export APIO_HOME_DIR=~/.icestudio/apio/
apio install scons
apio install ecp5@2019.12.11`

For now i can run 'verify' and 'build' for ice40 and ECP5 designs.

Probably far too experimental but i did not want to keep this to myself. ;-)

cavearr, 2021-07-27:
Thanks a lot!  Do you like make this for all of users? i could help you in the proccess, If you want to take this task, tell me!

janrinze, 2021-07-27:
> Thanks a lot! Do you like make this for all of users? i could help you in the proccess, If you want to take this task, tell me!

If we can get the dependencies fixed and have a script to detect aarch64 we might be able to automatically download the binaries from Leonard Laszlo's github releases.

For now I am busy figuring out why several packages are missing for aarch64.
It seems that the new travis build system of github is not configured to produce aarch64 binary releases for the APIO packages.

janrinze, 2021-07-27:
after some more searching it seems that building for ARM64 / aarch64 isn't straight forward using the automated build system of github. Apparently now Microsoft is in charge of github the whole stuff runs of windows azure systems. Building for Linux-x64 is supported and recently also OSX (both x86 and ARM64) but the maintainers mention that the servers don't support linux-aarch64. This is due to Microsoft cloud not supporting it.

janrinze, 2021-07-27:
Uploading to the lattice ICE40 breakout board works after the following commands:
```
export APIO_HOME_DIR=~/.icestudio/apio/ 
apio install system@1.1.1
```

Obijuan, 2021-07-27:
Great news! The system tools package is deprecated. The new one is [tools-oss-cad-suite](https://github.com/FPGAwars/tools-oss-cad-suite/wiki). It still does not have the packages for aarch64 because I am working on the migration, but they will

cavearr, 2021-07-27:
Hi Janrinze!  i'm studing your work and try to reproduce to view the possibe task and organize the work because we need to port first apio to arm64 and then icestudio.

tell us if you need some help.

Obijuan, 2021-07-27:
I've already added the system package for the aarch64. It would be great if you could try it:

```
apio install oss-cad-suite
```

Then try the following commands for checking if the package work:

```
apio system --lsusb
apio system --lsftdi
```

janrinze, 2021-07-28:
```
$ apio install oss-cad-suite
(DEBUG) Profile path: /home/janrinze/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/janrinze/.icestudio/apio
Installing oss-cad-suite package:
Download tools-oss-cad-suite-linux_aarch64-0.0.5.tar.gz
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Package 'oss-cad-suite' has been successfully installed!
```

```
apio system --lsusb
(DEBUG) Profile path: /home/janrinze/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/janrinze/.icestudio/apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin/lsusb
/home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin/lsusb: line 6: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/lib/ld-linux-aarch64.so.1: No such file or directory
```

janrinze, 2021-07-28:
Same error with libftdi:
```
apio system --lsftdi
(DEBUG) Profile path: /home/janrinze/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/janrinze/.icestudio/apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin/lsftdi
/home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin/lsftdi: line 6: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/lib/ld-linux-aarch64.so.1: No such file or directory
```

Obijuan, 2021-07-28:
Thanks for the test.  I copied the wrong library. Now it should be fixed. Can you please try again?
```
apio install -f oss-cad-suite
```
-f is for forcing the installation, as you already have the version installed

Then:
```
apio system --lsftdi
```

janrinze, 2021-07-29:
```
$ apio system --lsftdi
(DEBUG) Profile path: /home/janrinze/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/janrinze/.icestudio/apio
(DEBUG) Run Command: lsftdi
(DEBUG) System_base_dir: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/janrinze/.icestudio/apio/packages/tools-oss-cad-suite/bin/lsftdi
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: FER-RADIONA-EMARD, Description: ULX3S FPGA 85K v3.0.8
````

looks correct now.

closed 2021-08-03 by Obijuan

cavearr, 2021-08-28:
Hi @janrinze , now we have arm 64 icestudio build! you could download from actions or from https://downloads.icestudio.io

Thanks for show us the way!

jr-peterzon, 2021-10-01:
APIO still has some trouble to install the proper versions when using the nightly build. manual installing the apio dependencies works and then the whole icestudio app works.