FPGAwars/icestudio #528: lsusb 'PosixPath' object is not iterable

AlienTux, 2021-08-14:
Hi there.

I'm trying to run Icestudio on Solus 4.3, but the upload button returns a list of errors in a red sort of debug temporary window.

Had to take a screenshot because I didn't know how to select the text or find the debug file.  This here is the text:

```
(DEBUG) Porfile path: /home/kurt/.icestudio/apio/profile.json
(DEBUG) Home_dir:/home/kurt/.icestudio/apio
(DEBUG) Profile path: /home/kurt/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/kurt/.icestudio/apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: /home/kurt/.icestudio/apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/kurt/.icestudio/apio/packages/tools-oss-cad-suite/lsusb 'PosixPath' object is not iterable
```

Is there any way I can fix this? Or is it an issue in Icestudio itself?

Thank you!

cavearr, 2021-08-14:
Hi Kurt!, i'm trying to reproduce your problem, i'm installing Solus 4.3 and tell you asap.

AlienTux, 2021-08-18:
Veo que eres de españa! Soy de guatemala, muchas gracias! Quedo pendiente que quiero usar Icestudio para dar unas clases :) perdón que use el comentario para cosas sin sentido, pero me quería presentar.

Obijuan, 2022-05-02:
Creo que esta issue está solucionada. ¿Puedes probar la última wip? La puedes descargar desde aquí:

https://downloads.icestudio.io/

cavearr, 2024-02-05:
Cierro la issur por falta de respuesta y su antigüedad, la damos por solucionada.

closed 2024-02-05 by cavearr