FPGAwars/icestudio #529: Installing icestudio for all users on Windows

hpsteidle, 2021-08-17:
I want to use icestudio (a great tool) in our school. We use Windows10 Edu clients. Every time, when a student logs in, there will be created a new local user with a predefined configuration.
This behaviour causes the problem, that every time, a student logs in, the Toolchain for icestudio must be newly installed. This needs a lot of time. Also after some testinstallations i got an error from github because of too much requests per hour from our school-IP-address. 
Is there a posibility to install icestudio or the toolchain for all users on a Win10-system?  Alternatively is there a way to install apio-packages from a downlowded repository (offline-installation)?

cavearr, 2021-08-17:
Hi! i'm trying to  fix it or look for a solution.

TimRudy, 2023-02-20:
This would be a great contribution, probably real easy, for someone who wants to start in open source!
I checked where to start...

`this.installOnlineApio = function`
- Check `"pip install"`
- Check and use Icestudio's command output feature (View menu) - In fact when I open command output, I see info that almost gives away the whole game:
`"set APIO_HOME_DIR="C:\Users\xyz\.icestudio\apio"& set PATH=C:\Users\xyz\.icestudio\venv\Scripts;"`
- `ENV_DIR`, virtual environment directory, in the source code is relevant

Thanks anyone for any contribution!

Obijuan, 2023-02-20:
> Also after some testinstallations i got an error from github because of too much requests per hour from our school-IP-address.

This problem with the github api has been solved in apio 0.8.2 (still not released). It will be availabe in the next Icestudio stable version

cavearr, 2025-01-19:
I'm closing the issue because i think this is fixed, if is needed reopen it.

closed 2025-01-19 by cavearr