FPGAwars/icestudio #530: Exporting Verilog code of subblock changes subblock

BPJWES, 2021-08-17:
When exporting a subblock as verilog (requiring the user to be in the edit mode): 
[image]

Icestudio changes the subblock ports to "FPGA ports" and converts the subblock to effectively a standalone block. This occurs when going from the edit mode ( the lock symbol is unlocked) to back to the read-only mode (the lock symbol should lock again).

[image]

The lock symbol disappears, and the program enters an unusable state as the original top level module disappears. This can possibly lead to a significant loss of work.

cavearr, 2021-08-17:
Thanks for your feedback, i'm try to reproduce this bug and fix it thanks!

cavearr, 2021-09-03:
Hi! please check if in the last WIP this is fixed!, i'm waiting for your news.

Thanks!

BPJWES, 2021-09-06:
Platform: Windows 8.1
Version Icestudio: w202109060709

I can still reproduce the bug in the latest WIP

TimRudy, 2024-04-18:
If this bug exists, it will be directed related to: https://github.com/FPGAwars/icestudio/issues/651, so I will find out during testing and will fix it

cavearr, 2024-04-18:
Thanks Tim! i'm checking it , have you find a pattern to reproduce?

TimRudy, 2024-04-18:
No I haven't looked at it. Just reviewing the Issues list :-)

cavearr, 2025-01-19:
I think this is fixed in the last wip https://downloads.icestudio.io , if anyone said nothing in some days i'll close it.