FPGAwars/icestudio #531: Feature request: Verilog Code Blocks from external .v files

BPJWES, 2021-08-17:
It would be really nice to be able to use an external editor for verilog code blocks, as these can be rather large in which case it is sometimes easier to edit the code externally.

I made a small UI mockup which could be a way to use this functionality.

[image]

cavearr, 2021-08-17:
Hi! your idea is very welcome and converge with new features that i'm working in code blocks.

Thanks a lot for your mockup and your feedback!! i'm maintain open this issue to tell you about the progress of it.

cavearr, 2025-01-19:
Hi! we develop this feature , in other way but in very similar, in a few days the feature will evolve to link files on disk instead of import if the user wants with a acheck.

If you want to read more, here the discussion: https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/W9oHHQCxnQA/m/O6wCY53HDwAJ?utm_medium=email&utm_source=footer

This is available in the last wip https://downloads.icestudio.io

closed 2025-01-19 by cavearr