FPGAwars/icestudio #533: ULX3S (85f?) IO connections change upon loading etc.

BPJWES, 2021-08-23:
Version Info:
Icestudio Version: 0.7.0
OS: Windows 8.1

When connecting certain inputs and outputs for the ULX3S 85F (in my case, but is probably similar for other models of the ULX3S), the connections change when loading and changing the  "name"of the connection.

I have made a minimalist example demonstrating the problem, I have connected all the IO pins which I have found which demonstrate this problem, and connected them as shown in the image below:
[image]

If the project is now saved, the program closed, restarted, and the project reopened, we get the result as shown in the image below.
[image]

It turns out that the same problem occurs when changing the name of an in- or output block. For example changing >GN1 to something else will change the IO pin it is connected to.

Right now I manually fix these changes each time I open a project, but this is somewhat error prone.

I have attached the project which demonstrates this problem (in a zipped format, so github allows me to attach it to the bug)
[testfile_for_IO_bug_ULX3S_85f.zip](https://github.com/FPGAwars/icestudio/files/7035172/testfile_for_IO_bug_ULX3S_85f.zip)

jojo535275, 2021-08-23:
Hello,

This may be the same issue discussed here https://github.com/FPGAwars/icestudio/issues/499

in short, the problem is that icestudio mixes the pins when **all the characters** in the name of one pin are also included in the name of another pin.

in your case,  :    "gn1"  is included "**gn1**9"  , "gp1"  is included "**gp1**9", ....

A workaround (that i am similarly using for my colorlight i5 board) is to modify the pinout.json file  of you ulx3 board ( in folder ...\icestudio\resources\boards\ulx3s-85f )

by renaming pins :

"gn1" --> "gn1_"
"gp1"  --> "gp1_"
"gn2"  --> "gn2_"
"gp2"  --> "gp2_"

BPJWES, 2021-08-24:
I think the issue you mentioned seems very similar
Thank you, the workaround you mentioned does seem to work well.

It would nevertheless be nice if a proper fix for this issue could be found.