FPGAwars/icestudio #542: What would be right way to use PLL

goran-mahovlic, 2021-10-09:
Hi to all, thanks again on this amazing tool, I am still learning how to use it, but now I have some useful samples for workshop.

But I have tried to get pll as block, and now I am in trouble. 

[image]

And got error: Unknown module type: EHXPLLL

I am trying to add GPDI output, and I created all needed blocks except PLL

[image]

cavearr, 2021-10-10:
I am researching the problem and found the error but I am thinking how to fix it (at least it is one step).

The problem lies in a verilog statement that I think iverilog supports us, or at least not with the default Apio flags.

I'm working on it, I'll tell you ASAP!
[image]
[image]
[image]

goran-mahovlic, 2021-10-10:
> what Icestudio version are you using?

[image]

I am actually copy pasting samples from @mmicko apio workshop so maybe we could grab flags from there

https://github.com/ulx3s/fpga-odysseus

stas2k, 2022-08-04:
I have the same issue when using EHXPLLL in an ECP5 design. Build works, but Verify does not.

```
iverilog -B "/home/stas/.icestudio/apio/packages/tools-oss-cad-suite/lib/ivl" -o hardware.out -D VCD_OUTPUT= -D NO_INCLUDES "/home/stas/.icestudio/apio/packages/tools-oss-cad-suite/share/yosys/ecp5/cells_sim.v" PxsCourt.v PxsStrVGAJoin.v PxsVGAComp.v VGASyncGen.v main.v
VGASyncGen.v:62: error: Unknown module type: EHXPLLL
2 error(s) during elaboration.
*** These modules were missing:
EHXPLLL referenced 1 times.
***
scons: *** [hardware.out] Error 2
========================= [ ERROR ] Took 0.08 seconds =========================
```

cavearr, 2024-04-22:
with a lot of delay but @teaofjay fix it, the pr is merged into apio develop branch. thanks aganin, i'm closing this thread, but if the error appears reopen it. 

Thanks @teaofjay !