FPGAwars/icestudio #549: Adding custom boards

Synthron, 2021-11-02:
Is it possible to add custom board definitions so I can use my own FPGA boards with it?

benitoss, 2021-11-10:
Yes, lately I have added new boards.  Example are Colorlight boards in the ECP5 section
The FPGA of the new board has to be supported by Apio toolchain.

John-Westlake, 2021-11-20:
I've opened a new issue - but it appears there is no way to define the FPGA boards clock frequency, with iCEStudio / nextpnr defaulting to 12MHz...

Supported iCE40 boards such as Alchitry CU iCE40-HX8K and iCE40HX8K-EVB feature a 100MHz FPGA clock, however it appears that iCEStudio does not recognize the selected Boards actual clock frequency - and defaults to 12MHz.

When nextpnr is invoked by iCEStudio its target frequency is set to a "default" of 12MHz - resulting in timing issues when using higher speed FPGA clocks (such as the 100MHz fitted as standard on Alchitry CU iCE40-HX8K and iCE40HX8K-EVB boards etc).

After little digging, the it appears a default nextpnr Clock Constraint can be applied by --freq {freq} (in MHz), however I have no idea how to achieve this when iCEStudio invokes nextpnr (I cannot modify the nextpnr "command line").

I really would appreciate some urgent help please, as I currently cannot complete my project with the timing issues as a result of the default "too low" 12MHz clock - is there an easy method (under Windows8 / iCEStudio version 0.7.1w202110210410) to set a 100MHz FPGA clock?

Ultimately, it be great to have an option in iCEStudio under say the "Select" tab to choose Default or user selectable FPGA frequency - for those who have modify there iCE40 FPGA boards or whose Boards have higher clock frequency's then the default 12MHz.

John-Westlake, 2021-11-20:
Also, it be great to be great if iCEStudio supported the iCE40 FPGA family without "Dev Boards" - so the iCE40 family & package type (BG121 / CB132 / CT256 etc) could be selected much the same way as the current "Boards" are selected, but - with the option to select programmer type and target FPGA clock frequency.

The Lattice iCEProg programmer (as used on the Lattice iCE40-HX8K Breakout Board etc.) seems to be one of the best options to program  "Custom" FPGA boards / designs....

factoidforrest, 2025-01-22:
Hi there, I know this is an old issue but I'm curious if it ever went anywhere. There still seems to be a real lack of integrated and easy to use FPGA tools. I am building custom hardware but would love to use icestudio

cavearr, 2025-01-22:
Hi @factoidforrest ! now if you want to integrate new board, you need to integrate in apio and then in iceStudio.

But very soon (in a month or something near this) i'm releasing a new way to integrate only in Icestudio, i'm telling you in this thread.