FPGAwars/icestudio #550: Parameter in the block is allways 0

benitoss, 2021-11-10:
In the last nightly version 0.7.1w202110210410 there is a bug in the parameter part.
When you synthetize the code, the parameter always takes the value 0

Obijuan, 2021-11-10:
It wouled be great if you could provide an icestudio file with a minimal circuit with that bug. It is for undertanding better the bug and testing it in other boards.

benitoss, 2021-11-10:
In this example file 
[tm1638_led_key__test.zip](https://github.com/FPGAwars/icestudio/files/7515462/tm1638_led_key__test.zip)

IceStudio synthetizes but I got this error
main.v:3498: Warning: Range [-1:-1] select out of bounds on signal `\divcounter': Setting 1 LSB bits to undef.
Please, Could you explain why I got this error ???

Thanks

jojo535275, 2021-11-13:
Ola Fernando,

- i tryed your file wit my icestudio 0.7.1w202109100309  with apio 0.7.6, and also got the exact same warning.
- it is a warning not an error so maybe  we can just ignore it !?
- i created the smallest ice file(see joined file) where this warning appears:

[range warning.zip](https://github.com/FPGAwars/icestudio/files/7532467/range.warning.zip)
[image]
Yosis gives the warning in command output:
  yosys -p "synth_ecp5 -json hardware.json" -q main.v
  main.v:30: Warning: Range [-1:-1] select out of bounds on signal `\pres': 

and if we export the verilog code with the corresponding icestudio function we have

[image]
- So, the problem is that icestudio affects (at line 25 ) by default 0 as value for the external parameter (above a block). 
and at line 30 & 31  we have a N-1 (0-1) and yosis  sees that as potential problem.  It is looking for errors at each individual module/block, module by module.
-As we are quite often using N-1 in our blocks verilog code, if we want to decrease the number of yosis warnings we may change the default value given for a parameter in a block (eg: to 1 instead of 0 ).

benitoss, 2021-11-15:
I think the warning is because the internal parameter N in the instance  module is 0 and when we call the instance we set the parameter with a value greater than 1 , so finally the Bitfile generated perfectly  works. It is only a warning. 
Thanks fort checking it

cavearr, 2021-11-24:
Closing the issue, reopen if it is necesary.

closed 2021-11-24 by cavearr