FPGAwars/icestudio #551: Does not support Linux distros with python3.10

Min1123, 2021-11-17:
On Fedora 35, /usr/bin/python3.10 is the default installed via the DNF package manager.

IceStudio will not install the stable toolchain (from 0.6.1, 0.7.0, and a 0.7.1w nightly on Linux x64, appimage and linux ZIP), it says "At least Python 3.5 is required" and will not attempt the install.

This check does not appear to respect the Edit->Preferences->Python Environment settings.

Naheel-Azawy, 2022-01-20:
Hi,
Looks like the issue is clearly here: https://github.com/FPGAwars/icestudio/blob/057e4c8b0eb3ad5c48eddd0f91cc1d01199f75e7/app/scripts/services/utils.js#L120
I think checking could be as simple as parsing the version number and checking that's it is greater than 3.7 or whatever.

cavearr, 2022-01-24:
Hi! could you download the last wip and test if it works with python 3.10?

The last wip you could download from:

https://downloads.icestudio.io

If not work, tell me your os version please to try to replicate it.

Thanks!

Naheel-Azawy, 2022-01-25:
@cavearr Just tested and it seems to be working now. Thanks.
But as a little suggestion, why not just parse the version number instead of hardcodding the checks for each version?
Otherwise, every time we get a new version of python the application must be updated and the code will keep getting unnecessarily bigger.

I mean, instead of what we have here:
https://github.com/FPGAwars/icestudio/blob/master/app/scripts/services/utils.js#L149
we can have something like:
```js
version = result.split(" ")[1].split(".")
return version[0] >= 3 && version[1] >= 5
```
Of course it's better to have some additional checks for safety. But that's what I just had in mind.

cavearr, 2022-05-09:
Hi Naheel! thanks for your feedback and sorry for the delay in my response! i'm working in core things and dont review the issues .

Thanks for your code suggestions, i'll take it for comming versions! thanks again and thanks for your feedback!

closed 2022-05-09 by cavearr