FPGAwars/icestudio #553: 0.6.0 to 0.7.1w202110210410 upgrade on Win7: "nw_elf.dll" not found

mj1911, 2021-11-20:
Hello, I tried upgrading the 0.6.0 release to the latest nightly.  While 0.7.1 did say it was going to uninstall 0.6.0, it did not.  As a consequence, not all of the files were installed properly and 0.7.1 would not run.  Even supplying the missing .dll just resulted in another error.

Solution was to uninstall 0.7.1 and reinstall just 0.7.1.

Perhaps this should be obvious for nightlies, but I think if the 0.7.1 installer alerts the user that 0.6.0 will be uninstalled first, then it should actually uninstall it, or stop installation for manual uninstall.

cavearr, 2021-11-24:
Thanks for your feedback!, i'm marking this as enhancement for future versions.

cavearr, 2024-02-05:
this issue is very old, this stuff is solved in the last wips . You can download from https://downloads.icestudio.io or from github actions.

closed 2024-02-05 by cavearr